libusb/libusb #550: segfault on libusb_init on fedora ws29, while using 1.0.22_rc4 and 1.0.22 itself

alxbur, 2019-04-04:
an attempt to initialise library with libusb_init cause segmentation fault on line 2139 of core.c	ctx = calloc(1, sizeof(*ctx) + priv_size); Any advice on how to fix the issue greatly appreciated.
OS: Fedora WS29, libusb:libusb-1.0.22-rc4
thanks a lot,
Alex

hjelmn, 2019-04-05:
Very odd. The line in question shouldn't cause any issues.

alxbur, 2019-04-05:
printed on terminal: malloc.c:2385: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
Aborted (core dumped)

hjelmn, 2019-04-05:
Looks like something else is to blame. Might be worth running under valgrind to see what's going wrong.

alxbur, 2019-04-05:
Just tested libusb-1.023-rc1 - same error- segfault, same place:calloc in core.c

hjelmn, 2019-04-05:
Can you provide a reproducer? Either that or valgrind output would be helpful. Can't reproduce this issue in an Ubuntu 19.04 VM.

alxbur, 2019-04-05:
Hjelmn, thanks for taking a look on the issue. I'm not so experienced programmer and do not know how to use valgrind. Below is terminal output:
[alexb@localhost ve12demo_fw2f]$ ./ve12demo_sw2f 
ve12demo_sw2f: malloc.c:2385: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.
Aborted (core dumped)

alxbur, 2019-04-05:
Nathan, I'm using Fedora 29 workStation, kernel:$ uname -r
5.0.5-200.fc29.x86_64
the code causing trouble:
libusb_context*   pusb_ctx;
ret =libusb_init(&pusb_ctx);

alxbur, 2019-04-05:
Nathan, here we are:

==18266== Memcheck, a memory error detector
==18266== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==18266== Using Valgrind-3.14.0-353a3587bb-20181007X and LibVEX; rerun with -h for copyright info
==18266== Command: ./ve12demo_sw2f
==18266== Parent PID: 13014
==18266== 
--18266-- 
--18266-- Valgrind options:
--18266--    --leak-check=full
--18266--    --show-leak-kinds=all
--18266--    --track-origins=yes
--18266--    --verbose
--18266--    --log-file=valgrind-out.txt
--18266-- Contents of /proc/version:
--18266--   Linux version 5.0.5-200.fc29.x86_64 (mockbuild@bkernel04.phx2.fedoraproject.org) (gcc version 8.3.1 20190223 (Red Hat 8.3.1-2) (GCC)) #1 SMP Wed Mar 27 20:58:04 UTC 2019
--18266-- 
--18266-- Arch and hwcaps: AMD64, LittleEndian, amd64-cx16-rdtscp-sse3-avx
--18266-- Page sizes: currently 4096, max supported 4096
--18266-- Valgrind library directory: /usr/lib64/valgrind
--18266-- Reading syms from /home/alexb/projects/ve12demo_sw2f/bin/Debug/ve12demo_sw2f
--18266-- Reading syms from /usr/lib64/ld-2.28.so
--18266-- Reading syms from /usr/lib64/valgrind/memcheck-amd64-linux
--18266--    object doesn't have a symbol table
--18266--    object doesn't have a dynamic symbol table
--18266-- Scheduler: using generic scheduler lock implementation.
--18266-- Reading suppressions file: /usr/lib64/valgrind/default.supp
==18266== embedded gdbserver: reading from /tmp/vgdb-pipe-from-vgdb-to-18266-by-alexb-on-localhost.localdomain
==18266== embedded gdbserver: writing to   /tmp/vgdb-pipe-to-vgdb-from-18266-by-alexb-on-localhost.localdomain
==18266== embedded gdbserver: shared mem   /tmp/vgdb-pipe-shared-mem-vgdb-18266-by-alexb-on-localhost.localdomain
==18266== 
==18266== TO CONTROL THIS PROCESS USING vgdb (which you probably
==18266== don't want to do, unless you know exactly what you're doing,
==18266== or are doing some strange experiment):
==18266==   /usr/lib64/valgrind/../../bin/vgdb --pid=18266 ...command...
==18266== 
==18266== TO DEBUG THIS PROCESS USING GDB: start GDB like this
==18266==   /path/to/gdb ./ve12demo_sw2f
==18266== and then give GDB the following command
==18266==   target remote | /usr/lib64/valgrind/../../bin/vgdb --pid=18266
==18266== --pid is optional if only one valgrind process is running
==18266== 
--18266-- REDIR: 0x401fa50 (ld-linux-x86-64.so.2:strlen) redirected to 0x580c37e2 (???)
--18266-- REDIR: 0x401f820 (ld-linux-x86-64.so.2:index) redirected to 0x580c37fc (???)
--18266-- Reading syms from /usr/lib64/valgrind/vgpreload_core-amd64-linux.so
--18266-- Reading syms from /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so
==18266== WARNING: new redirection conflicts with existing -- ignoring it
--18266--     old: 0x0401fa50 (strlen              ) R-> (0000.0) 0x580c37e2 ???
--18266--     new: 0x0401fa50 (strlen              ) R-> (2007.0) 0x0483bd40 strlen
--18266-- REDIR: 0x401c230 (ld-linux-x86-64.so.2:strcmp) redirected to 0x483ce20 (strcmp)
--18266-- REDIR: 0x401ffb0 (ld-linux-x86-64.so.2:mempcpy) redirected to 0x48408a0 (mempcpy)
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_xrc-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_webview-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_html-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_qa-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_adv-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_core-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_baseu_xml-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_baseu_net-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_baseu-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
[... 60 lines cut ...]
--18266-- Reading syms from /usr/lib64/libcairo-gobject.so.2.11600.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libatk-1.0.so.0.23009.1
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libatk-bridge-2.0.so.0.0.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwayland-client.so.0.3.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libepoxy.so.0.0.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libharfbuzz.so.0.10800.7
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libpangoft2-1.0.so.0.4200.3
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libfontconfig.so.1.12.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libfreetype.so.6.16.1
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libpcre.so.1.2.11
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libffi.so.6.0.2
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libselinux.so.1
--18266--    object doesn't have a symbol table
--182

alxbur, 2019-04-05:
I tried Daniel's listdevsin examples- runs well. the only difference I see is 
	r = libusb_init(NULL);
vs.
ret =libusb_init(&pusb_ctx);

hjelmn, 2019-04-05:
@alxbur The valgrind output got truncated. Can you put it in a gist (https://gist.github.com/)?

hjelmn, 2019-04-05:
Working fine for me:

```c
#include <stdlib.h>
#include <stdio.h>

#include <libusb.h>

int main (void) {
  libusb_context *context;

  libusb_init (&context);
  libusb_exit (context);

  return 0;
}
```

Tested macOS 10.14 and Ubuntu 16.04 (4.15 kernel). Not sure why it is failing for you.

alxbur, 2019-04-05:
==18266== Memcheck, a memory error detector
==18266== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==18266== Using Valgrind-3.14.0-353a3587bb-20181007X and LibVEX; rerun with -h for copyright info
==18266== Command: ./ve12demo_sw2f
==18266== Parent PID: 13014
==18266== 
--18266-- 
--18266-- Valgrind options:
--18266--    --leak-check=full
--18266--    --show-leak-kinds=all
--18266--    --track-origins=yes
--18266--    --verbose
--18266--    --log-file=valgrind-out.txt
--18266-- Contents of /proc/version:
--18266--   Linux version 5.0.5-200.fc29.x86_64 (mockbuild@bkernel04.phx2.fedoraproject.org) (gcc version 8.3.1 20190223 (Red Hat 8.3.1-2) (GCC)) #1 SMP Wed Mar 27 20:58:04 UTC 2019
--18266-- 
--18266-- Arch and hwcaps: AMD64, LittleEndian, amd64-cx16-rdtscp-sse3-avx
--18266-- Page sizes: currently 4096, max supported 4096
--18266-- Valgrind library directory: /usr/lib64/valgrind
--18266-- Reading syms from /home/alexb/projects/ve12demo_sw2f/bin/Debug/ve12demo_sw2f
--18266-- Reading syms from /usr/lib64/ld-2.28.so
--18266-- Reading syms from /usr/lib64/valgrind/memcheck-amd64-linux
--18266--    object doesn't have a symbol table
--18266--    object doesn't have a dynamic symbol table
--18266-- Scheduler: using generic scheduler lock implementation.
--18266-- Reading suppressions file: /usr/lib64/valgrind/default.supp
==18266== embedded gdbserver: reading from /tmp/vgdb-pipe-from-vgdb-to-18266-by-alexb-on-localhost.localdomain
==18266== embedded gdbserver: writing to   /tmp/vgdb-pipe-to-vgdb-from-18266-by-alexb-on-localhost.localdomain
==18266== embedded gdbserver: shared mem   /tmp/vgdb-pipe-shared-mem-vgdb-18266-by-alexb-on-localhost.localdomain
==18266== 
==18266== TO CONTROL THIS PROCESS USING vgdb (which you probably
==18266== don't want to do, unless you know exactly what you're doing,
==18266== or are doing some strange experiment):
==18266==   /usr/lib64/valgrind/../../bin/vgdb --pid=18266 ...command...
==18266== 
==18266== TO DEBUG THIS PROCESS USING GDB: start GDB like this
==18266==   /path/to/gdb ./ve12demo_sw2f
==18266== and then give GDB the following command
==18266==   target remote | /usr/lib64/valgrind/../../bin/vgdb --pid=18266
==18266== --pid is optional if only one valgrind process is running
==18266== 
--18266-- REDIR: 0x401fa50 (ld-linux-x86-64.so.2:strlen) redirected to 0x580c37e2 (???)
--18266-- REDIR: 0x401f820 (ld-linux-x86-64.so.2:index) redirected to 0x580c37fc (???)
--18266-- Reading syms from /usr/lib64/valgrind/vgpreload_core-amd64-linux.so
--18266-- Reading syms from /usr/lib64/valgrind/vgpreload_memcheck-amd64-linux.so
==18266== WARNING: new redirection conflicts with existing -- ignoring it
--18266--     old: 0x0401fa50 (strlen              ) R-> (0000.0) 0x580c37e2 ???
--18266--     new: 0x0401fa50 (strlen              ) R-> (2007.0) 0x0483bd40 strlen
--18266-- REDIR: 0x401c230 (ld-linux-x86-64.so.2:strcmp) redirected to 0x483ce20 (strcmp)
--18266-- REDIR: 0x401ffb0 (ld-linux-x86-64.so.2:mempcpy) redirected to 0x48408a0 (mempcpy)
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_xrc-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_webview-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_html-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_qa-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_adv-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_gtk3u_core-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_baseu_xml-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_baseu_net-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /usr/lib64/libwx_baseu-3.0.so.0.4.0
--18266--    object doesn't have a symbol table
--18266-- Reading syms from /opt/lib/libusb-1.0.so.1.1.0
--18266-- Reading syms from /usr/lib64/libudev.so.1.6.11
[... 1199 lines cut ...]
==18266==    by 0x46258F: main (ve12demo_sw2fApp.cpp:17)
==18266==  Address 0x1ebe0350 is 8 bytes after a block of size 5,272 alloc'd
==18266==    at 0x4838E86: operator new(unsigned long) (vg_replace_malloc.c:344)
==18266==    by 0x465886: ve12demo_sw2fFrame::ve12demo_sw2fFrame(wxWindow*, int) (ve12demo_sw2fMain.cpp:201)
==18266==    by 0x46262A: ve12demo_sw2fApp::OnInit() (ve12demo_sw2fApp.cpp:26)
==18266==    by 0x462B3E: wxAppConsoleBase::CallOnInit() (app.h:93)
==18266==    by 0x5449D31: wxEntry(int&, wchar_t**) (in /usr/lib64/libwx_baseu-3.0.so.0.4.0)
==18266==    by 0x46258F: main (ve12demo_sw2fApp.cpp:17)
==18266== 
==18266== Invalid write of size 8
==18266==    at 0x483F854: memset (vg_replace_strmem.c:1252)
==18266==    by 0x46B706: usbComm::usbComm(wxWindow*, wxStatusBar*) (usbComm.cpp:313)
==18266==    by 0x4658A9: ve12demo_sw2fFrame::ve12demo_sw2fFrame(wxWindow*, int) (ve12demo_sw2fMain.cpp:201)
==18266==    by 0x46262A: ve12demo_sw2fApp::OnInit() (ve12demo_sw2fApp.cpp:26)
==18266==    by 0x462B3E: wxAppConsoleBase::CallOnInit() (app.h:93)
==18266==    by 0x5449D31: wxEntry(int&, wchar_t**) (in /usr/lib64/libwx_baseu-3.0.so.0.4.0)
==18266==    by 0x46258F: main (ve12demo_sw2fApp.cpp:17)
[... cut ...] c'd
==18266==    at 0x4838E86: operator new(unsigned long) (vg_replace_malloc.c:344)
==18266==    by 0x465886: ve12demo_sw2fFrame::ve12demo_sw2fFrame(wxWindow*, int) (ve12demo_sw2fMain.cpp:201)
==18266==    by 0x46262A: ve12demo_sw2fApp::OnInit() (ve12demo_sw2fApp.cpp:26)
==18266==    by 0x462B3E: wxAppConsoleBase::CallOnInit() (app.h:93)
==18266==    by 0x5449D31: wxEntry(int&, wchar_t**) (in /usr/lib64/libwx_baseu-3.0.so.0.4.0)
==18266==    by 0x46258F: main (ve12demo_sw2fApp.cpp:17)
==18266==

alxbur, 2019-04-05:
Was not able to use gist, all controlls were greyed out, so I just put again valgrind out in comment

alxbur, 2019-04-05:
[strace_out.txt](https://github.com/libusb/libusb/files/3049923/strace_out.txt)
[valgrind-out.txt](https://github.com/libusb/libusb/files/3049925/valgrind-out.txt)

alxbur, 2019-04-05:
Thank you everybody for your expertise & time! Any clues, how to fix the issue at this point?
Alex.

hjelmn, 2019-04-05:
Was about to say just attach the output :).

This is what stands out:

```
valgrind: m_mallocfree.c:307 (get_bszB_as_is): Assertion 'bszB_lo == bszB_hi' failed.
valgrind: Heap block lo/hi size mismatch: lo = 5344, hi = 0.
This is probably caused by your program erroneously writing past the
end of a heap block and corrupting heap metadata.  If you fix any
invalid writes reported by Memcheck, this assertion failure will
probably go away.  Please try that before reporting this as a bug.
```

Valgrind noticed something writing past the end of an allocation. This can make subsequent calls to ```malloc()``` abort. I would look through the invalid writes and see if anything jumps out at you.

alxbur, 2019-04-07:
ctx = calloc(1, sizeof(ctx* ) + priv_size);
what should be a value of priv_size, gdb shows 0, is it correct value? ctv shown asc"as optimised out"- could it be part of the issue?

hjelmn, 2019-04-07:
context_priv_size is 0 for the Linux backend so that is expected. Try moving libusb_init to an earlier point in your program (such as at the beginning of main()). If that succeeds it is more evidence that you are probably stomping on the heap somewhere in your program.

alxbur, 2019-04-07:
Thsnks, hjelmn, I double checked, libusb initialisation done inusbcomm class constructor, which in turn called as a first line in wxwidgets constructor., therefore, libusbinit is the first thing happenned.
THank you for the advice anyway, will continue digging around..Any comments on non working example: listdevs? What I'm doing wrong?

alxbur, 2019-04-07:
Hjelmn, thank you so much, I found one place that potentially could corrupt heap. Now gdb shows diffwerent spot of segfault.Program received signal SIGABRT, Aborted.
In raise () (/lib64/libc.so.6)
#8  0x00007ffff71c96bb in check_usb_vfs (dirname=dirname@entry=0x7ffff71d16df "/dev/bus/usb") at os/linux_usbfs.c:292
/home/alexb/prereq/builds/libusb-1.0.23-rc1/libusb/os/linux_usbfs.c:292:9477:beg:0x7ffff71c96bb
At /home/alexb/prereq/builds/libusb-1.0.23-rc1/libusb/os/linux_usbfs.c:292
#8  0x00007ffff71c96bb in check_usb_vfs (dirname=dirname@entry=0x7ffff71d16df "/dev/bus/usb") at os/linux_usbfs.c:292
/home/alexb/prereq/builds/libusb-1.0.23-rc1/libusb/os/linux_usbfs.c:292:9477:beg:0x7ffff71c96bb
Program terminated with signal SIGABRT, Aborted.
Debugger finished with status 0
`static int check_usb_vfs(const char *dirname)
{
	DIR *dir;
	struct dirent *entry;
	int found = 0;

-->>	**dir = opendir(dirname);**
	if (!dir)
		return 0;

	while ((entry = readdir(dir)) != NULL) {
		if (entry->d_name[0] == '.')
			continue;

		/* We assume if we find any files that it must be the right place */
		found = 1;
		break;
	}`
at this point dirname=0x7ffff71d16df: "/dev/bus/usb", and dir somehow marked as optimised out
Could be threading be a leading  actor in the tragedy?

dickens, 2020-01-13:
Most definitely not a libusb issue.  Unfortunately the heap space of your program is corrupt.  Since this is not a libusb issue, I'm going to close this.

closed 2020-01-13 by dickens