libusb/libusb #570: make check fail

ryankumar, 2019-05-08:
=============================================================
1 of 8 tests did not behave as expected (1 unexpected passes)
=============================================================
Makefile:447: recipe for target 'check-TESTS' failed
make[2]: *** [check-TESTS] Error 1
make[2]: Leaving directory '/home/mahannee/packages/libusb/libusb-0.1.12/tests'
Makefile:547: recipe for target 'check-am' failed
make[1]: *** [check-am] Error 2
make[1]: Leaving directory '/home/mahannee/packages/libusb/libusb-0.1.12/tests'
Makefile:550: recipe for target 'check-recursive' failed
make: *** [check-recursive] Error 1

LudovicRousseau, 2019-05-08:
It works for me:
```
$ make check
Making check in libusb
make[1]: Nothing to be done for `check'.
Making check in doc
make[1]: Nothing to be done for `check'.
Making check in examples
make[1]: Nothing to be done for `check'.
Making check in tests
make[1]: Nothing to be done for `check'.
```

Please give mode details to reproduce the problem.

ryankumar, 2019-05-08:
what is mode details ??

LudovicRousseau, 2019-05-08:
Sorry, "more details".
Like a precise list of steps to reproduce the issue.

ryankumar, 2019-05-08:
mahannee@VTA076L:~/packages/libusb/libusb-0.1.12$ sudo make check
Making check in .
make[1]: Entering directory '/home/mahannee/packages/libusb/libusb-0.1.12'
make[1]: Leaving directory '/home/mahannee/packages/libusb/libusb-0.1.12'
Making check in tests
make[1]: Entering directory '/home/mahannee/packages/libusb/libusb-0.1.12/tests'
make  check-TESTS
make[2]: Entering directory '/home/mahannee/packages/libusb/libusb-0.1.12/tests'
Dev #1: Linux 4.19.6 xhci-hcd - xHCI Host Controller
Dev #1: Linux 4.19.6 xhci-hcd - xHCI Host Controller
  Dev #2: Logitech - Lenovo USB Optical Mouse
  Dev #3: NOVATEK - USB Keyboard
Dev #1: Linux 4.19.6 ehci_hcd - EHCI Host Controller
  Dev #2: 8087 - 8000
Dev #1: Linux 4.19.6 ehci_hcd - EHCI Host Controller
  Dev #2: 8087 - 8008
PASS: testlibusb
bus/device  idVendor/idProduct
004/001     163841d6b/40000003
- Unable to fetch manufacturer string
- Unable to fetch product string
  wTotalLength:         31
  bNumInterfaces:       1
  bConfigurationValue:  1
  iConfiguration:       0
  bmAttributes:         e0h
  MaxPower:             0
    bInterfaceNumber:   0
    bAlternateSetting:  0
    bNumEndpoints:      1
    bInterfaceClass:    9
    bInterfaceSubClass: 0
    bInterfaceProtocol: 0
    iInterface:         0
      bEndpointAddress: 81h
      bmAttributes:     03h
      wMaxPacketSize:   4
      bInterval:        12
      bRefresh:         0
      bSynchAddress:    0
003/003     40000461/40000010
- Unable to fetch manufacturer string
- Unable to fetch product string
  wTotalLength:         59
  bNumInterfaces:       2
  bConfigurationValue:  1
  iConfiguration:       0
  bmAttributes:         a0h
  MaxPower:             50
    bInterfaceNumber:   0
    bAlternateSetting:  0
    bNumEndpoints:      1
    bInterfaceClass:    3
    bInterfaceSubClass: 1
    bInterfaceProtocol: 1
    iInterface:         0
      bEndpointAddress: 81h
      bmAttributes:     03h
      wMaxPacketSize:   8
      bInterval:        10
      bRefresh:         0
      bSynchAddress:    0
    bInterfaceNumber:   1
    bAlternateSetting:  0
    bNumEndpoints:      1
    bInterfaceClass:    3
    bInterfaceSubClass: 0
    bInterfaceProtocol: 0
    iInterface:         0
      bEndpointAddress: 82h
[... 187 lines cut ...]
1b: 

XPASS: hub_strings
bus/device  idVendor/idProduct
004/001     1D6B/0003    driver: hub
003/003     0461/0010    driver: usbhid
003/003     0461/0010    driver: usbhid
003/002     17EF/6019    driver: usbhid
003/001     1D6B/0002    driver: hub
002/002     8087/8000    driver: hub
002/001     1D6B/0002    driver: hub
001/002     8087/8008    driver: hub
001/001     1D6B/0002    driver: hub
XPASS: driver_name
=============================================================
3 of 8 tests did not behave as expected (3 unexpected passes)
=============================================================
Makefile:447: recipe for target 'check-TESTS' failed
make[2]: *** [check-TESTS] Error 1
make[2]: Leaving directory '/home/mahannee/packages/libusb/libusb-0.1.12/tests'
Makefile:547: recipe for target 'check-am' failed
make[1]: *** [check-am] Error 2
make[1]: Leaving directory '/home/mahannee/packages/libusb/libusb-0.1.12/tests'
Makefile:550: recipe for target 'check-recursive' failed
make: *** [check-recursive] Error 1

LudovicRousseau, 2019-05-08:
You are using libusb-0.1.12.
The latest version here is 1.0.22. Please upgrade.
Or try libusb-compat https://github.com/libusb/libusb-compat-0.1

closed 2019-05-08 by LudovicRousseau