libusb/libusb #572: macOS: "Crashed Thread: org.libusb.device-hotplug" whenever Google Chrome is quit?

ylluminate, 2019-05-20:
I'm having a peculiar error whenever I quit Google Chrome: [Problem Report.txt](https://dqdpzeak1aol0.cloudfront.net/items/413b3w462P1P0Y3T3L1f/Process%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20Google%20C.txt)

Notice:
```
Crashed Thread:        23  org.libusb.device-hotplug

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
```
And the crashed thread in question:
```
Thread 23 Crashed:: org.libusb.device-hotplug
0   com.apple.CoreFoundation      	0x00007fff50b20f4b __CFRunLoopServiceMachPort + 549
1   com.apple.CoreFoundation      	0x00007fff50b203da __CFRunLoopRun + 1619
2   com.apple.CoreFoundation      	0x00007fff50b1fb35 CFRunLoopRunSpecific + 459
3   com.apple.CoreFoundation      	0x00007fff50b1f945 CFRunLoopRun + 40
4   com.google.Chrome.framework   	0x000000011449d784 0x10fc39000 + 75908996
5   libsystem_pthread.dylib       	0x00007fff7ccdc2eb _pthread_body + 126
6   libsystem_pthread.dylib       	0x00007fff7ccdf249 _pthread_start + 66
7   libsystem_pthread.dylib       	0x00007fff7ccdb40d thread_start + 13
```

Any idea of what might be going on here?

LudovicRousseau, 2019-05-20:
Maybe Chrome forgot to call `libusb_exit()` before exiting?

seanm, 2020-03-20:
Chrome could have a customized libusb also, you should report Chrome crashes to google, not here.

LudovicRousseau, 2020-03-21:
I may also add that I can't reproduce the problem on macOS 10.14.6 with Chrome 80.0.3987.149

Also your "Problem Report.txt" file does not list any libusb library loaded.

seanm, 2020-08-09:
We should probably close this...

mcuee, 2020-08-11:
Close for now. Please re-open if same issues pop up.

closed 2020-08-11 by mcuee