libusb/libusb #573: [RFE] User error messages

victortoso, 2019-05-21:
Hi,

In spice-gtk we have been using a wrapper around libusb_error codes that gives a more user friendly message instead, [spice_usbutil_libusb_strerror](https://gitlab.freedesktop.org/spice/spice-gtk/blob/master/src/usbutil.c#L61).

My suggestion in this issue is to create API in libusb with better error description or perhaps improve [libusb_error_name](https://github.com/libusb/libusb/blob/master/libusb/core.c#L2682) to achieve a more user friendly message.

freddy77, 2019-05-21:
Just call `libusb_strerror` ? Or it's a problem of i18n ? I would just print the number and the English version (the one from the `libusb_strerror`).

victortoso, 2019-05-28:
libusb_strerror sounds fine indeed, I missed that.

closed 2019-05-28 by victortoso