libusb/libusb #574: LNK2001: unresolved external symbol __imp__iob

pateldeep73241, 2019-05-22:
in my visual studio 2019 project i am using libusb-1.0.lib(core.obj) to communicate PC with usb device, and i see this error
libusb-1.0.lib(core.obj) : error LNK2001: unresolved external symbol __imp__iob

cant resolve this error... can anyone help me fix it.

I imported an 2008 visual studio project (it works in here !!) into 2019 visual studio.

pateldeep73241, 2019-05-22:
i added this and it got fixed 
#pragma comment(lib, "legacy_stdio_definitions.lib")
#ifdef __cplusplus
FILE iob[] = { *stdin, *stdout, *stderr };
extern "C" {
    FILE * __cdecl _iob(void) { return iob; }
}
#endif

pateldeep73241, 2019-05-22:
But now when i debug/run my code, it gives an exception... 
Unhandled exception at 0x00007FFD2874B6E0 (ucrtbased.dll) in <executable_name>.exe: 0xC0000005: Access violation reading location 0x00007FFD28C6A5B8.

Not sure what is this.. Can anyone help ?

mcuee, 2019-06-24:
Please use libusb mailing list for question like this. This seems to have nothing to do with libusb.

liugangnhm, 2020-03-25:
Same Issue.

i used `libusb-1.0.23` windows static lib with visual studio 2019. It seems to be because of different visual studio version, but i dont know how to fix.

xyx0826, 2020-08-11:
I ran into the same issue today. My solution is to recompile libusb with VS2019. The recompiled static library works for me.

mcuee, 2020-08-11:
Just re-poen this issue in case people still face the same problem.

Take not the pre-compiled binary is provided as it is. You can always re-compile if it does not work for you.

dickens, 2020-12-11:
This cannot be fixed, unfortunately.  Mixing objects compiled with different versions of Visual Studio, or not using the same runtime library selection, is ultimately going to fail.

In the future Windows binaries archive I will include objects compiled for each of the supported Visual Studio versions.

closed 2020-12-11 by dickens