libusb/libusb #575: Unable to claim interface under OSX Mojave

simonhsays, 2019-06-01:
I have a device driver which utilizes libusb. It has worked great on previous versions of OSX, but under Mojave it seems I can no longer claim_interface. It keeps failing during the call to IOCreatePlugInInterfaceForService(), returning an "unkown error (0xe00002c7)", which I have tracked down to an IOKit "unsupported function" error.

Interestingly, after the initial call to IOCreatePlugInInterfaceForService during the libusb_claim_interface routine, I always see a "notifying context xyz of device disconnect", and think this might be related. However, my hotplug callbacks do not register a device detachment/attachment.

Here's an example of the output:

[ 0.005074] [00000307] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x7fa460c07c20 with handle 1
[ 0.005125] [00000307] libusb: debug [libusb_hotplug_register_callback] new hotplug cb 0x7fa460c07bd0 with handle 2
[ 0.005128] [00000307] libusb: debug [libusb_get_device_list] 
8 Devices in list.
[ 0.005175] [00000307] libusb: debug [libusb_get_device_descriptor] 
 Found the device

[ 0.005181] [00000307] libusb: debug [libusb_get_device_descriptor] 
Number of possible configurations: 1  Device Class: 255  VendorID: 1240  ProductID: 63062
[ 0.005191] [00000307] libusb: debug [libusb_get_config_descriptor] index 0
Interfaces: 1 ||| Number of alternate settings: 1 | Interface Number: 0 | Number of endpoints: 4 | Descriptor Type: 5 | EP Address: 1 | Descriptor Type: 5 | EP Address: 129 | Descriptor Type: 5 | EP Address: 2 | Descriptor Type: 5 | EP Address: 130 | 

[ 0.005215] [00000307] libusb: debug [libusb_open] open 20.32
[ 0.005293] [00000307] libusb: debug [darwin_open] device open for access
[ 0.005301] [00000307] libusb: debug [libusb_kernel_driver_active] interface 0
[ 0.005334] [00000307] libusb: debug [libusb_get_configuration] 
[ 0.005350] [00000307] libusb: debug [libusb_get_configuration] active config 0
[ 0.005355] [00000307] libusb: debug [libusb_set_configuration] configuration 1
[ 2.215913] [00000307] libusb: debug [libusb_get_configuration] 
[ 2.215929] [00000307] libusb: debug [libusb_get_configuration] active config 1
[ 2.215938] [00000307] libusb: debug [libusb_claim_interface] interface 0
[ 2.216373] [00002807] libusb: debug [darwin_devices_detached] notifying context 0x7fa460c05640 of device disconnect
[ 2.216823] [00000307] libusb: error [darwin_claim_interface] IOCreatePlugInInterfaceForService: unknown error (0xe00002c7)
 Could not claim interface!
LIBUSB_ERROR_OTHER
[ 2.216832] [00000307] libusb: debug [libusb_close] 
[ 2.216863] [00000307] libusb: warning [darwin_close] USBDeviceClose: no connection to an IOService
[ 2.216886] [00000307] libusb: debug [libusb_exit] 
[ 2.216888] [00000307] libusb: debug [libusb_exit] destroying default context

hjelmn, 2019-07-08:
What version of libusb are you using?

simonhsays, 2019-07-11:
I tried with 1.0.22 as well as 1.0.23rc with no luck.

I ended up rolling back the OSX IOUSBFamily.kext to a previous version
(900.4.1 from El Capitan) and this solved my problem. However, this is not
an ideal solution.

Simon

kajakang, 2019-09-11:
How did you roll back the IOUSBFamily.kext? I have similar problem for Mojave.

simonhsays, 2019-11-30:
FYI. This issue seems to have been resolved in OSX Catalina.

closed 2019-11-30 by simonhsays

hjelmn, 2019-11-30:
Well, that is rather annoying. That says it could be a bug in Mojave. Would be nice to find a workaround if one exists. There were a number of nasty macOS bugs when IOUSBFamily was rewritten and we were able to work around some of them.

lwei2, 2020-10-16:
> Well, that is rather annoying. That says it could be a bug in Mojave. Would be nice to find a workaround if one exists. There were a number of nasty macOS bugs when IOUSBFamily was rewritten (10.10 I think, been awhile now) and we were able to work around some of them.

Hi, hielmn.How do I send SCSI commands to USB devices on MacOS？