libusb/libusb #576: missing install-filter-win.exe

seginming, 2019-06-03:
I have a Masoon power monitor devices which comes with a python SDK to measure realtime power usage through USB cable.

The Monsoon python library uses libusb which is suggested to be downloaded at The Monsoon python library uses. To use these SDK/API's on windows platform, The additional step of installing the libusb filter is necessary. 

However, only 32bit support is there. When i try to install this filter, i keep getting below error:
This is a 64bit operating system and requires the 64bit libusb-win32 installer application.

Is there a 64bit version of libusb filter installer available?
Please help.

seginming, 2019-06-03:
btw, the 32bit version is downloaded at https://sourceforge.net/projects/libusb-win32/

mcuee, 2019-06-24:
Closed. libusb-win32 is a separate project.
And you should use libusbK or Zadig for the task as mentioned in libusb-win32 project file download area.

closed 2019-06-24 by mcuee