libusb/libusb #579: unable to compile on centos 7

slavamas, 2019-06-12:
ran ./configure.
result:
checking whether gcc and cc understand -c and -o together... yes
checking operating system... Linux
checking for library containing clock_gettime... none required
checking libudev.h usability... yes
checking libudev.h presence... yes
checking for libudev.h... yes
checking for udev_new in -ludev... no
configure: error: udev support requested but libudev not installed

LudovicRousseau, 2019-06-13:
> udev support requested but libudev not installed

=> install libudev-devel package

closed 2019-06-13 by LudovicRousseau

slavamas, 2019-06-13:
I was not able to find package libudev
however there is 
libgudev
Installed Packages
Name        : libgudev1
Arch        : x86_64
Version     : 219
Release     : 62.el7_6.6
Size        : 50 k
Repo        : installed
From repo   : updates
Summary     : Libraries for adding libudev support to applications that use glib
URL         : http://www.freedesktop.org/wiki/Software/systemd
License     : LGPLv2+
Description : This package contains the libraries that make it easier to use libudev
            : functionality from applications that use glib.
is that the one needed?
yum list installed | egrep *udev
libgudev1.x86_64                        219-62.el7_6.6                 @updates 
libgudev1-devel.x86_64                  219-62.el7_6.6                 @updates 
python-pyudev.noarch                    0.15-9.el7                     @anaconda
system-config-printer-udev.x86_64       1.4.1-21.el7                   @base

hjelmn, 2019-06-13:
@slavamas centos moved it to systemd-devel. I have never been a fan of RedHat-style distros and this sort of thing doesn't help :).

slavamas, 2019-06-13:
thank you @hjelmn - I am aware of it and agree it was not very helpful/convenient . :)
looks like it compiled now (with bunch of warnings though)

Thank you!