libusb/libusb #580: 【HELP！！！】libusb_control_transfer returns -1

Sleepingbug, 2019-06-13:
1，test environment：
windows 10 x64, libusb win32, visual studio 2017
lib-1.0.23

2，issue:
libusb_control_transfer returns -1.
I use a tool named Bus Hound to send the same data to my usb device, my device can receive the data and the data can be captured by Bus Hound. 

3, my source code:
`#pragma once

#include "libusb.h"
#include <stdio.h>
#include <sys/types.h>
#include <string.h>
#include <windows.h>
#include <errno.h>

#define VID 0x0E8D
#define PID 0x2005

int main()
{
	libusb_device_handle *devh = NULL;
 
	int ret = libusb_init(NULL);
	if (ret < 0) {
		perror("libusb_init");
		return ret;
	}

	libusb_set_debug(NULL, LIBUSB_LOG_LEVEL_DEBUG );

	devh = libusb_open_device_with_vid_pid(NULL, VID, PID);

	if (!devh) {
		printf("libusb_open_device_with_pid_vid Error: %d,\n", ret);
		Sleep(1000);
		libusb_exit(NULL);
	}
	
	Sleep(1000);
	// claim interface
	int interface_number = 4;
	ret = libusb_claim_interface(devh, interface_number);

	if (ret < 0) {
		printf("libusb_claim_interface Error: %d,\n", ret);
		Sleep(2000);
		if (devh) {
			libusb_close(devh);
			devh = NULL;
		}
		return ret;
	}

	unsigned char type = 0x21;
	unsigned char req = 0x09;//HID_SET_REPORT
	uint16_t wValue = 0x0201;
	uint16_t wIndex = 0x01;
	uint16_t length16 = 0x0010;// data length
	unsigned char data[16] = {  0x01, 0x00, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03 };

	printf("---before---%d\n",GetTickCount());
	//FAILED, -1, libusb_error; 
	// both 3 and 0x83 are failed, i want to find the correct one.
	for (wIndex = 0; wIndex < 10; ++wIndex) {
		ret = libusb_control_transfer(devh, type, req, wValue, wIndex, data, length16, 10000);
		printf("libusb_control_transfer,wIndex=%d, ret=%d\n", wIndex,ret );
	}
[... 525 lines cut ...]
6	bInterfaceSubClass	1	42h	
7	bInterfaceProtocol	1	01h	
8	iInterface	1	4Ch	"My Interface"
Endpoint Descriptor 04 4 Out, Bulk, 64 bytes
Offset	Field	Size	Value	Description
0	bLength	1	07h	
1	bDescriptorType	1	05h	Endpoint
2	bEndpointAddress	1	04h	4 Out
3	bmAttributes	1	02h	Bulk
	1..0: Transfer Type		......10 	Bulk
	7..2: Reserved		000000.. 	
4	wMaxPacketSize	2	0040h	64 bytes
6	bInterval	1	00h	
Endpoint Descriptor 84 4 In, Bulk, 64 bytes
Offset	Field	Size	Value	Description
0	bLength	1	07h	
1	bDescriptorType	1	05h	Endpoint
2	bEndpointAddress	1	84h	4 In
3	bmAttributes	1	02h	Bulk
	1..0: Transfer Type		......10 	Bulk
	7..2: Reserved		000000.. 	
4	wMaxPacketSize	2	0040h	64 bytes
6	bInterval	1	00h	

`

Sleepingbug, 2019-06-13:
This problem took me 3 days but I haven't solved it yet. It makes me crazy.
 T_T

closed 2019-06-14 by Sleepingbug

nireus1997, 2021-06-17:
@Sleepingbug Did you solve it? i have a same question. thanks

mcuee, 2021-06-17:
Try using HIDAPI and not libusb for HID device.
https://github.com/libusb/hidapi

mcuee, 2021-06-17:
Also please read HID specifications here.
https://www.usb.org/hid

nireus1997, 2021-06-18:
`	buf[0] = 0x0;
	buf[1] = 0x21;
	buf[2] = 0x09;
	buf[3] = 0x00;
	buf[4] = 0x02;
	buf[5] = 0x00;
	buf[6] = 0x00;
	buf[7] = 0x01;
	buf[8] = 0x00;
	res = hid_write(handle, buf, 9);
	if (res < 0) {
		printf("Unable to write()\n");
		printf("Error: %ls\n", hid_error(handle));
	}
`
According to your prompt, I tried this operation, but the result still returns 'invalid function'。 i tried google\stackoverflow..  but nothing help.
[image]

any other suggesstions? thanks

mcuee, 2021-06-18:
You have to know the communication protocol of the device. If not, no one can help you,

nireus1997, 2021-06-18:
sorry, my english is poor。
first step: 
i use bus hound to send a report like this:(this works)
[image]
then, second step:
i want use libusb or hidapi(Recommended by you) to send the same data to hid device。
but libusb_control_transfer(libusb) or hid_write(hidapi)  getlasterror() return 1 (invalid function).

mcuee, 2021-06-18:
Your English perfectly okay. For technical support like this, please use libusb-devel mailing list. Thanks.