libusb/libusb #581: why wIndex is different between libusb0 and libusb1 ?

Sleepingbug, 2019-06-14:
libusb0:
wIndex = 0x04;
ret = usb_control_msg(devh, 0x21,0x09, 0x0202, wIndex, data, data_length, 1000);

libusb1:
wIndex = 0x05;
ret = libusb_control_transfer(devh, 0x21,0x09, 0x0202, wIndex, data, data_length, 10000);

why wIndex is different between libusb0 and libusb1 ?

hjelmn, 2019-06-14:
Can you give more info? wIndex is identical in libusb 0.1.x and libusb-1.0. Are you referring to libusb0 (windows)?  That is a completely different project.

Sleepingbug, 2019-06-14:
Testing environment:

Sleepingbug, 2019-06-14:
Testing environment:
windows 10 , visual studio 2017 ,x64, C/C++

hjelmn, 2019-06-14:
That isn't enough info. Were you using libusb0 (https://sourceforge.net/projects/libusb-win32/)? If so that has never been affiliated with libusb. They implemented their own version of the libusb API.

Sleepingbug, 2019-06-14:
Thank you very much!

closed 2019-06-14 by Sleepingbug