libusb/libusb #584: on the android 8 platfrom,libusb_init is ok，but libusb_get_device_list return 0

laomengandroid, 2019-06-17:
Use the sample you provided，Step by step；
1.added log on the testlibusb.c adb libusb Source code
2.will binaryed libusb1.0 Library push   /system/lib64
3.will binaryed testlibusb  push  /system/bin
4.adb shell  /system/bin/ testlibusb
5.view log printf:
   >libusb_init is successfull
        r = libusb_init(ctx);
	@printf("libusb_init==%d",r);   printf r ==0;
   >libusb_get_device_list return 0
     cnt = libusb_get_device_list(NULL, &devs);
	  if (cnt < 0)
		return (int)cnt;
        @printf("libusb_get_device_list==%d\n",cnt); ---> cnt==0

wbing-xu, 2019-06-20:
I also found the same problem.
when i  reset to tag 'v1.0.22',  device can be found!

hjelmn, 2019-06-20:
What version are you running?

cnlohr, 2019-08-23:
It's broken in 1.0.23. I can't find any code path which can actually do enumeration on android in 1.0.23.

sicauyanglei, 2019-10-21:
What can I do for this problem?

sicauyanglei, 2019-10-22:
> ag 'v1.0.22

how can I do it

cnlohr, 2019-10-22:
use `git checkout` to specify the older branch.  I.e. `git checkout v1.0.22` in the folder where you've checked out the source.

mcuee, 2020-01-06:
https://github.com/libusb/libusb/commit/2f3bc98b0d0f4766496df53c855685a5f0e5e7cf

Maybe this commit has some problems.

mcuee, 2020-01-06:
#233 comment is relevant.

laran002, 2020-01-09:
Parts of 2f3bc98, related to linux_scan_devices(), is reverted back in #637

IXMSbhatt, 2020-05-12:
libusb_get_device_list still return 0 in android 7.1.1 too

cnlohr, 2020-05-12:
Just checking - are you running on a rooted device with selinux set to not enforce?

IXMSbhatt, 2020-05-12:
it's rooted device and selinux is not set. With libusb-1.0.22 it's working perfectly fine. I have revert back 1.0.23 to 1.0.22. Now my problem is solved with libusb-1.0.22.

green-green-avk, 2020-05-19:
Hmm...
What do you think about this approach: <https://github.com/green-green-avk/LibUsbManager> as it uses <https://github.com/green-green-avk/libusb-1.0.23-android-helper-service-patch>?

Device enumeration and hot plug/unplug events are supported.
No existing software (if dynamically linked to **libusb**) recompilation are required (in case of some Linux distributition under **PRoot**).
Tested as a part of <https://github.com/green-green-avk/AnotherTerm>.

hongguangli, 2020-07-14:
Thanks to Almighty github, and everyone on github.(translate.google 0_0)

mcuee, 2020-07-15:
So there seems to be a regression from 1.022 to 1.023. For those who are interested in Android, root cause analysis and patches are welcome.

mcuee, 2020-08-03:
https://github.com/libusb/libusb/pull/760

Not so sure if the above will help.

mcuee, 2020-12-19:
Please try 1.0.24 to see if it is okay in 1.0.24. Thanks.

mcuee, 2021-02-05:
@IXMSbhatt and other users, please help to check if 1.0.24 release sorted out the issue. Thanks.

mcuee, 2021-02-26:
I assume the issues have been sorted out in 1.0.24. Close for now. Please reopen if this is not sorted out.

And for Android users, please refer to Pull Requests #830 and #874.

Related discussions:
https://github.com/libusb/libusb/issues/717
https://github.com/libusb/libusb/issues/868

adhiti33, 2021-07-02:
@mcuee 

Usb on android is not working for me. I am using latest v1.0.24. Can you confirm below understanding is right ?

1> https://github.com/libusb/libusb/pull/830/commits/ff03b23022782dfaf4bfbac54c6251aea4e5e613

>> Individually picking this patch did not work for me

2> https://github.com/libusb/libusb/pull/874

>> Do I need to directly pick libusb git from @xloem distribution ??

mcuee, 2021-07-02:
Please comment on #830 if it does not work for you. Please try out #874.

mcuee, 2021-07-02:
@adhiti33 BTW, does libusb-1.0.22 works for you or not? If it does not, then your issue is different from this one. Thanks.

adhiti33, 2021-07-02:
@mcuee We can resolve this issue. I will discuss further in https://github.com/libusb/libusb/pull/874 and try it.

closed 2022-02-06 by mcuee