libusb/libusb #585: [handle_bulk_completion] low level error -71

coder2yuan, 2019-06-17:
Dear: 
I use libusb transfer some datas, but on some PC(ubuntu 14, linux 3.11) get errors, that is the log:
[149.815683] [00004fe3] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f36f803ffa8 has callback 0x50af60
[149.815704] [00004fe3] libusb: debug [sync_transfer_cb] actual_length=8192
[149.815725] [00004fe3] libusb: debug [libusb_free_transfer] transfer 0x7f36f803ffa8
[149.815750] [00004fe3] libusb: debug [libusb_alloc_transfer] transfer 0x7f36f803ffa8
[149.815769] [00004fe3] libusb: debug [libusb_submit_transfer] transfer 0x7f36f803ffa8
[149.815789] [00004fe3] libusb: debug [add_to_flying_list] arm timerfd for timeout in 20000ms (first in line)
[149.815811] [00004fe3] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 8192
[149.815842] [00004fe3] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[149.815861] [00004fe3] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[149.816098] [00004fe3] libusb: debug [handle_events] poll() returned 1
[149.816141] [00004fe3] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=8192
[149.816162] [00004fe3] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[149.816181] [00004fe3] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[149.816200] [00004fe3] libusb: debug [disarm_timerfd] 
[149.816220] [00004fe3] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f36f803ffa8 has callback 0x50af60
[149.816239] [00004fe3] libusb: debug [sync_transfer_cb] actual_length=8192
[149.816260] [00004fe3] libusb: debug [libusb_free_transfer] transfer 0x7f36f803ffa8
[149.816284] [00004fe3] libusb: debug [libusb_alloc_transfer] transfer 0x7f36f803ffa8
[149.816302] [00004fe3] libusb: debug [libusb_submit_transfer] transfer 0x7f36f803ffa8
[149.816321] [00004fe3] libusb: debug [add_to_flying_list] arm timerfd for timeout in 20000ms (first in line)
[149.816342] [00004fe3] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 8192
[149.816372] [00004fe3] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[149.816391] [00004fe3] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[149.816621] [00004fe3] libusb: debug [handle_events] poll() returned 1
[149.816664] [00004fe3] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=8192
[149.816685] [00004fe3] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[149.816704] [00004fe3] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[149.816723] [00004fe3] libusb: debug [disarm_timerfd] 
[149.816742] [00004fe3] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f36f803ffa8 has callback 0x50af60
[149.816762] [00004fe3] libusb: debug [sync_transfer_cb] actual_length=8192
[149.816782] [00004fe3] libusb: debug [libusb_free_transfer] transfer 0x7f36f803ffa8
[149.816806] [00004fe3] libusb: debug [libusb_alloc_transfer] transfer 0x7f36f803ffa8
[149.816825] [00004fe3] libusb: debug [libusb_submit_transfer] transfer 0x7f36f803ffa8
[149.816843] [00004fe3] libusb: debug [add_to_flying_list] arm timerfd for timeout in 20000ms (first in line)
[149.816864] [00004fe3] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 8192
[149.816894] [00004fe3] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[149.816913] [00004fe3] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[149.817006] [00004fe3] libusb: debug [handle_events] poll() returned 1
[149.817028] [00004fe3] libusb: debug [reap_for_handle] urb type=3 status=-71 transferred=1024
[149.817036] [00004fe3] libusb: debug [handle_bulk_completion] handling completion status -71 of bulk urb 1/1
[149.817045] [00004fe3] libusb: debug [handle_bulk_completion] low level error -71
[149.817052] [00004fe3] libusb: debug [disarm_timerfd] 
[149.817061] [00004fe3] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f36f803ffa8 has callback 0x50af60
[149.817069] [00004fe3] libusb: debug [sync_transfer_cb] actual_length=1024
[149.817078] [00004fe3] libusb: debug [libusb_free_transfer] transfer 0x7f36f803ffa8
 libusb: debug [sync_transfer_cb] actual_length=8192

hjelmn, 2019-07-08:
71 is EPROTO. you will likely need to dig into the USB driver or look for additional errors in dmesg to track this down. it isn't necessarily a libusb issue.

mcuee, 2020-01-07:
Close this for now. 
To OP: please reopen if you have new info.

closed 2020-01-07 by mcuee