libusb/libusb #586: Solaris backend depends on Solaris private symbols

kentsuji, 2019-06-18:
Solaris backend depends on the following double-linked list operation functions from libcmdutils:
list_insert_tail, list_create, list_remote_head, list_head, list_next

However, they are private symbols in libcmdutils library and moreover libcmdutils library is also a private library. libusb-1 should not directly refer those symbols. Due to this issue, linking a program with libusb-1 fails on the SPARC platform.  For example, building examples on SPARC fails as the following:

$ gmake
  CC       listdevs.o
  CCLD     listdevs
Undefined                       first referenced
 symbol                             in file
list_insert_tail                    ../libusb/.libs/libusb-1.0.so
list_create                         ../libusb/.libs/libusb-1.0.so
list_remove_head                    ../libusb/.libs/libusb-1.0.so
list_head                           ../libusb/.libs/libusb-1.0.so
list_next                           ../libusb/.libs/libusb-1.0.so
ld: fatal: symbol referencing errors
collect2: error: ld returned 1 exit status
gmake: *** [Makefile:421: listdevs] Error 1

Please note libusb-1 on x86 does not show this problem because libusb-1 is explicitly linked with libdevinfo library that is internally linked with libcmdutils. So, it's accidentally working on x86.
Looking at os/sunos_usb.c, I don't think it needs to use the double-linked list operations to build the argument list for execv.  Please see attached sample fix for this issue.  I kept the function structure as the same as possible.  But, since the number of the arguments to execv is not variable but always 5 in sunos_{detach,attach}_kernel_driver(), that could become much simpler.

Thanks,
Kenjiro

[sunos_usb.c.patch.txt](https://github.com/libusb/libusb/files/3299418/sunos_usb.c.patch.txt)

Mno-hime, 2019-09-27:
This should be fixed with https://github.com/libusb/libusb/pull/627.

dickens, 2020-01-13:
Looks like this was resolved, so closing.

closed 2020-01-13 by dickens