libusb/libusb #588: Mouse causing existing device should share parent error

yratof, 2019-06-25:
Having an issue where my SteelSeries Mouse is causing this reoccurring error:

```
[2019-06-25 12:11:54.738]: ClientConnection::start - Starting client connection id 0
[2019-06-25 12:11:54.740]: ClientConnection::send_connection_info - Sending connection id to client 0
libusb: error [init_device] device '\\.\USB#VID_1038&PID_1810&MI_01#7&2BE7A914&0&0001' is no longer connected!
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [init_device] device '\\.\USB#VID_1038&PID_1810&MI_01#7&2BE7A914&0&0001' is no longer connected!
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
libusb: error [windows_get_device_list] program assertion failed - existing device should share parent
```

This is all in relation to trying to get PSMOVESERVICE working

mcuee, 2019-06-28:
This should not be a real problem for your device, right?
Probably this may need to reduce to a warning level and not error in the log.

diabolo38, 2019-07-04:
what usb host controller?  This kind of log is quite common with some usb 3 xhci

mcuee, 2019-12-07:
Close this for now. The OP can reopen if providing more info.

closed 2019-12-07 by mcuee

brodiedundee, 2019-12-12:
I have the exact same issue any ideas i should try out? this occurs as soon as i start it ive tried on non usb3 ports it lets me get to stage 5/7 then back to 4/7 pairing the controller