libusb/libusb #589: Raspberry Pi receive fail when run second time. Reconnect USB

lee8871, 2019-07-08:


hjelmn, 2019-07-08:
Need a lot more details. What libusb version? What code is failing? etc..

lee8871, 2019-07-08:
Hello everyone, 

I have create a CMAKE project which connect to a libusb device.  It work well in Win10 and ubuntu.
But when I port it to my raspberry pi 3B and 3B+(I have 2 RasPis). It is running well when first running and fail for second. I have to reconnect USB cable than it run well at once. 
Sometimes, the project can run more than one time, may be 2 to 6 times, than fail.

Following is version for libusb in raspi: this libusb is installed by apt, I have not build it yet.
```
pi@raspberrypi:~ $ apt show libusb-1.0-0
Package: libusb-1.0-0
Version: 2:1.0.21-1
Priority: optional
Section: libs
Source: libusb-1.0
Maintainer: Aurelien Jarno <aurel32@debian.org>
Installed-Size: 119 kB
Depends: libc6 (>= 2.17), libudev1 (>= 183)
Homepage: http://www.libusb.info
Download-Size: 48.2 kB
APT-Manual-Installed: yes
APT-Sources: http://mirrors.tuna.tsinghua.edu.cn/raspbian/raspbian stretch/main armhf Packages
Description: userspace USB programming library
Library for programming USB applications without the knowledge
of Linux kernel internals.
.
This package contains what you need to run programs that use this
library.
```
I have tried to location error.  My log show the receive function return no 0 
``` c++
				if (LIBUSB_SUCCESS != libusb_bulk_transfer(mainDH, (1 + 0x80), pkg->u8, 31 + 3, &rcvd_len, 10)) {
					delete pkg;
					logger.infoPrint('U',"recv Pkg Timeover");
					return fail;
				}
```
I close all handle when process exit :
```c++
			int closeUsb() {
				if (mainDH != nullptr) {
					libusb_release_interface(mainDH, 0);
					libusb_close(mainDH);
					mainDH = nullptr;
				}
				if (mainDEV != nullptr) {
					libusb_unref_device(mainDEV);
					mainDEV = nullptr;
				}
				if(mainCTX!=nullptr){
					libusb_exit(mainCTX);					
					logger.infoPrint('U',"close mainCTX!");
					OsSupport::msSleep(50);
					mainCTX=nullptr;
				}				
				return done;
			}
```
I have try to reopen libusb device, I call  closeUsb() when reopen:
```c++
						if(access_reset_counter == ACCESS_RESET_MAX){
							logger.crashPrint("Access timeover too many times!");
							throw "Access timeover too many times!";
						}
						else{
							OsSupport::msSleep(50);
							int rev = openUsbByName(last_name);
							OsSupport::msSleep(50);
							if(rev == done){     
								logger.errPrint("Access timeover reset all USB%d done(recv:%d,send:%d), portname is %s"
								,access_reset_counter,recv_error_counter,send_error_counter, last_name);				
[... 91 lines cut ...]
[SRB LOG][1970-01-01 08:05:46]:
[SRB LOG][1970-01-01 08:05:51]:
[SRB LOG][1970-01-01 08:05:56]:
[SRB LOG][1970-01-01 08:06:01]:
	[SRBERR][]:Access timeover 0 (recv:5,send:0) 
	[SRBERR][]:Access timeover 1 (recv:5,send:0) 
	[SRBERR][]:Access timeover 2 (recv:5,send:0) 
	[SRBERR][]:Access timeover 3 (recv:5,send:0) 
	[SRBERR][]:Access timeover 4 (recv:5,send:0) 
[SRB LOG][1970-01-01 08:06:04]:
	[SRBERR][]:Access timeover 0 (recv:5,send:0) 
	[SRBERR][]:Access timeover 1 (recv:5,send:0) 
	[SRBERR][]:Access timeover 2 (recv:5,send:0) 
	[SRBERR][]:Access timeover 3 (recv:5,send:0) 
	[SRBERR][]:Access timeover 4 (recv:5,send:0) 
[SRB LOG][1970-01-01 08:06:08]:
[SRB LOG][1970-01-01 08:06:13]:
[SRB LOG][1970-01-01 08:06:18]:
```
In though test, my test skills and log system is not so good. But before I increase it, the issue CAN NOT REPRODUCE!(why) I JUST add log and USB reopen, and  reopen never triggered.
In my log shows that the issue less and less until disappears.

Now I report the issue. It may helpful if anyone encounter the same phenomenon.

Looking forward to your reply.

lee8871, 2019-07-08:
In this Log, when process run,record [SRB LOG][1970-01-01 08:06:08]: 
[SRBERR][]:Access timeover 1 (recv:5,send:0) 
“recv:5,” means receive function call fail 5 times `if (LIBUSB_SUCCESS != libusb_bulk_transfer(mainDH, (1 + 0x80), pkg->u8, 31 + 3, &rcvd_len, 10)) ` not the function return.

mcuee, 2019-12-07:
Looks not a libusb problem but rather a raspberry pi or device problem. Close this for now.

closed 2019-12-07 by mcuee