libusb/libusb #591: Bulk Tranfert Error In Payload [Win 7 - 10]

AurelienDelbart, 2019-07-10:
Hi,
I've been working on USB communication using libusb, in C++ with the Qt framework. It works fine on Linux, but on windows 7 and 10, I've got some trouble with the bulk transfer. On windows, I use the driver WinUsb, set on the device (atmelv71) with Zadig.
First of all, at the start of the transfert, this errors pops up in the console :
libusb: warning [winusbx_claim_interface] auto-claimed interface 0 (required to claim 1 with WinUSB)
libusb: error [do_close] Device handle closed while transfer was still being processed, but the device is still connected as far as we know
libusb: error [do_close] A cancellation hasn't even been scheduled on the transfer for which the device is closing
libusb: warning [check_pollfds] too many HANDLEs to wait on
libusb: warning [check_pollfds] too many HANDLEs to wait on
libusb: warning [check_pollfds] too many HANDLEs to wait on
libusb: warning [check_pollfds] too many HANDLEs to wait on
libusb: warning [check_pollfds] too many HANDLEs to wait on

After monitoring the data that is send on the device, I can see that they are corrupted. The payload data are fill up with 0xfffe. (this payload have been monitored with a software "Device Monitoring Studio").
But if I put the thread to sleep for 10 ms after each packet send (512bytes) It work fine, the send data are not corrupted.

Here is the sending code that doesn't work:

```c++
   if ((f_DeviceHandle == nullptr) || (f_DeviceIsAlive == false))
      return false;

   const Word PacketSize(f_UniversalPortSettings.GetEndpointOutPacketSize(portInfo.toInt()));
   const Byte EndpointOut(f_UniversalPortSettings.GetEndpointOut(portInfo.toInt()) | LIBUSB_ENDPOINT_OUT);

   // On découpe notre message à envoyer si celui-ci est plus grand que la taille maximale autorisée sur l'endpoint
   const int NbMessages((dataToSend.size() / PacketSize) + 1);

   for (int IndexMessage(0); IndexMessage < NbMessages; ++IndexMessage)
   {
      QByteArray DataToSend(dataToSend.mid((IndexMessage * PacketSize), PacketSize));

      libusb_transfer* WriteTransfert = libusb_alloc_transfer(0);
      libusb_fill_bulk_transfer(WriteTransfert, f_DeviceHandle, EndpointOut, reinterpret_cast<uchar*>(DataToSend.data()), DataToSend.size(), WriteTransfertCallBack, this, 0);
      int ReturnCode(libusb_submit_transfer(WriteTransfert));

      // On vérifie le code de retour
      if (ReturnCode < LIBUSB_SUCCESS)
         AwTracesLevelError(AW_TRACE_SRC_INFO % QStringLiteral("Failed to submit transfert at the endpoint \"") % QString::number(EndpointOut) % QStringLiteral("\" - Error code: ") % QString(libusb_error_name(ReturnCode)));
   }

   return true;
```

Here is the sending code that work :

```c++
   if ((f_DeviceHandle == nullptr) || (f_DeviceIsAlive == false))
      return false;

   const Word PacketSize(f_UniversalPortSettings.GetEndpointOutPacketSize(portInfo.toInt()));
   const Byte EndpointOut(f_UniversalPortSettings.GetEndpointOut(portInfo.toInt()) | LIBUSB_ENDPOINT_OUT);

   // On découpe notre message à envoyer si celui-ci est plus grand que la taille maximale autorisée sur l'endpoint
   const int NbMessages((dataToSend.size() / PacketSize) + 1);

   for (int IndexMessage(0); IndexMessage < NbMessages; ++IndexMessage)
   {
      QByteArray DataToSend(dataToSend.mid((IndexMessage * PacketSize), PacketSize));

      libusb_transfer* WriteTransfert = libusb_alloc_transfer(0);
      libusb_fill_bulk_transfer(WriteTransfert, f_DeviceHandle, EndpointOut, reinterpret_cast<uchar*>(DataToSend.data()), DataToSend.size(), WriteTransfertCallBack, this, 0);
      int ReturnCode(libusb_submit_transfer(WriteTransfert));

      QThread::currentThread()->msleep(10);

      // On vérifie le code de retour
      if (ReturnCode < LIBUSB_SUCCESS)
         AwTracesLevelError(AW_TRACE_SRC_INFO % QStringLiteral("Failed to submit transfert at the endpoint \"") % QString::number(EndpointOut) % QStringLiteral("\" - Error code: ") % QString(libusb_error_name(ReturnCode)));
   }

   return true;
```

Thanks.

diabolo38, 2019-07-13:
Just by a quick look i guess you use asynchronous API but are not waiting anywhere for completion likely freeing the request and related memory  before completed why the corrupted data ?
synchronous api may be simpler.
under window they are limited   number of xfer flying smthg like  64 as far i can remenber but ilay be wrong.
last it is not any wise and any  efficient to send many small message over bulk just  send less message but bigger the os and above layer will do better job that you may do here  unless tackling hi perf it wan't make muche difference to your app  a 1M tp 2M   bulk size per xfer is granted to work over all os ?

AurelienDelbart, 2019-07-13:
I've got a Callback called WriteTransfertCallBack, wich free the transfert and return an error if the tranfer status is not LIBUSB_SUCCESS. But, yes, I've tried with the synchronous way, and it worked. It is a little bit slower although. I've although solved the autoclaiming issue by asking on windows the first interface.

dickens, 2020-01-28:
Please try this again with the latest code.  The `too many HANDLEs to wait on` message should no longer be an issue.

Note that there is definitely an issue with your code closing the device before all transfers have completed.  This is a recipe for disaster.

mcuee, 2021-01-23:
Close this item due to inactivity. Please reopen if the latest libusb codes still have the issue.

closed 2021-01-23 by mcuee