libusb/libusb #594: Windows XP 32bit use libusb and UsbDk not support hub hot-plugging？

BruceMok, 2019-07-13:
**system architecture**：
              +-----------+                  +-----------+      
                virt-viewer   < ------- >    spice-gtk  
              +-----------+                  +-----------+ 
                     |
                     |
+------------+------------+ 
     libusb        libusbredir 
+------------+----------- -+ 
                     |
                     |
        +-----------------+
                 UsbDk 
        +-----------------+   

**issue**
When i plug in a hub, it work well and redirect. Then i plug out this hub, virt-viewer GUI blocking and have no response. I add some log in libusb source code, it seems to blocking at "windows_usbdk.c":
```
static void usbdk_close(struct libusb_device_handle *dev_handle)
{
        ......
	if (!usbdk_helper.StopRedirect(priv->redirector_handle)) {
		usbi_err(HANDLE_CTX(dev_handle), "Redirector shutdown failed");
	}
       ......
}
```

This issues 100% appears on Windows XP 32bit， but never appears on Win7 or Win10. Log as below：

> (remote-viewer.exe:3180): GSpice-WARNING **: notify_dev_state_change remove
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: notify_dev_state_change remove
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: 【USB】 RMV ENEVT!
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: 【USB】 RMV 1 spice_usb_device_manager_remove_dev
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: 【USB】 RMV 2 disconnect_device_sync 02e463f0
> [67.250000] [00000308] libusb: debug [libusb_get_bus_number] core.libusb_get_bus_number
> [67.250000] [00000308] libusb: debug [libusb_get_bus_number] core.libusb_get_bus_number
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: 【USB】 RMV 3 disconnect device from usb channel 042e7368
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: 【yanke】 disconnect_device 1
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: 【yanke】 disconnect_device 2
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: 【yanke】 disconnect_device 3
> [67.250000] [00000308] libusb: debug [libusb_release_interface] interface 0
> [67.250000] [00000308] libusb: debug [libusb_release_interface] interface 1
> [67.250000] [00000308] libusb: debug [libusb_release_interface] interface 2
> [67.250000] [00000308] libusb: debug [libusb_release_interface] interface 3
> [67.250000] [00000308] libusb: debug [libusb_reset_device] 
> [67.250000] [00000308] libusb: error [usbdk_reset_device] ResetDevice failed: [2] 系统找不到指定的文件。
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328] core.libusb_close start!
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328]libusb_close --- 0
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328]libusb_close --- 1
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328]libusb_close --- 2
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328]libusb_close --- 3
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328]ibusb_close --- 4
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328]libusb_close --- 5
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328]libusb_close --- 6
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328]libusb_close --- 7
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328]libusb_close --- 8
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328]libusb_close --- 9
> [67.250000] [00000e44] libusb: debug [handle_events] poll() returned 1
> [67.250000] [00000e44] libusb: debug [handle_events] caught a fish on the event pipe
> [67.250000] [00000e44] libusb: debug [handle_events] someone is closing a device
> [67.250000] [00000e44] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
> [67.250000] [00000e44] libusb: debug [libusb_try_lock_events] someone else is closing a device
> [67.250000] [00000308] libusb: debug [libusb_close] [10:40:14.328]libusb_close --- 10
> [67.250000] [00000e44] libusb: debug [libusb_event_handler_active] someone else is closing a device
> [67.250000] [00000e44] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
> [67.250000] [00000308] libusb: debug [usbdk_close] 10:40:14.328 yanke:usbdk_close start...
> [127.250000] [00000e44] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
> [127.250000] [00000e44] libusb: debug [libusb_try_lock_events] someone else is closing a device
> [127.250000] [00000e44] libusb: debug [libusb_event_handler_active] someone else is closing a device
> [127.250000] [00000e44] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: session: disconnecting 0
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: [spice_session_init] Supported channels: main, display, inputs, cursor, playback, record, usbredir, webdav
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: ==== set_property, auto-connect filter set to 0x03,-1,-1,-1,0|-1,-1,-1,-1,1
> 
> (remote-viewer.exe:3180): GSpice-WARNING **: 【USB-DEVICE-MANAGER】 spice_usb_device_manager_initable_init
> [138.062500] [000009e0] libusb: debug [libusb_init] core.libusb_init=== start!
> [138.062500] [000009e0] libusb: debug [libusb_init] libusb v1.0.22.11312
> [138.062500] [000009e0] libusb: debug [libusb_init] core.libusb_init=== list_add
> [138.062500] [000009e0] libusb: debug [libusb_init] core.libusb_init=== usbi_backend.init
> [138.062500] [000009e0] libusb: debug [windows_init] ------------------windows_nt_common.windows_init start
> [138.062500] [000009e0] libusb: debug [windows_init] windows_nt_common.windows_init FOCUS init_count=2
> [138.062500] [000009e0] libusb: debug [windows_init] ------------------FOCUS usbdk_backend.init START r=-99
> [138.062500] [000009e0] libusb: debug [usbdk_init] FOCUS windows_usbdk.usbdk_init... start
> [138.062500] [000009e0] libusb: debug [windows_init] ------------------FOCUS usbdk_backend.init [usbdk_available = true]
> [138.062500] [000009e0] libusb: debug [windows_init] ------------------FOCUS usbdk_backend.init END r=0
> [138.062500] [000009e0] libusb: debug [windows_init] ------------------windows_nt_common.windows_init end
> [138.062500] [000009e0] libusb: debug [usbi_add_pollfd] add fd 4 events 1
> [138.062500] [000009e0] libusb: debug [libusb_init] core.libusb_init=== end
> [138.062500] [000009e0] libusb: debug [libusb_get_device_list] core.libusb_get_device_list
> [138.062500] [000009e0] libusb: debug [libusb_get_device_list] 
> [138.062500] [000009e0] libusb: debug [usbdk_get_device_list] usbdk_get_device_list ====================
> [138.093750] [000009e0] libusb: debug [winusb_get_device_list] allocating new device for session [1AC0]
> [138.093750] [000009e0] libusb: debug [usbi_connect_device] core.usbi_connect_device
> [138.093750] [000009e0] libusb: debug [winusb_get_device_list] allocating new device for session [1AFC]
> [138.093750] [000009e0] libusb: debug [usbi_connect_device] core.usbi_connect_device
> [138.093750] [000009e0] libusb: debug [winusb_get_device_list] allocating new device for session [1BD4]
> [138.093750] [000009e0] libusb: debug [usbi_connect_device] core.usbi_connect_device
> [138.093750] [000009e0] libusb: debug [winusb_get_device_list] allocating new device for session [1C24]
> [138.093750] [000009e0] libusb: debug [usbi_connect_device] core.usbi_connect_device
> [138.093750] [000009e0] libusb: debug [get_api_type] driver(s): WinUSB
> [138.093750] [000009e0] libusb: debug [get_api_type] matched driver name against WinUSB
> [138.093750] [000009e0] libusb: debug [winusb_get_device_list] allocating new device for session [8F4]
> [138.093750] [000009e0] libusb: debug [usbi_connect_device] core.usbi_connect_device
> [138.093750] [000009e0] libusb: debug [get_api_type] driver(s): HidUsb
> [138.093750] [000009e0] libusb: debug [get_api_type] matched driver name against HID API
> [138.093750] [000009e0] libusb: debug [winusb_get_device_list] allocating new device for session [1538]
> [138.093750] [000009e0] libusb: debug [usbi_connect_device] core.usbi_connect_device
> [138.093750] [000009e0] libusb: debug [get_api_type] driver(s): usbccgp
> [138.093750] [000009e0] libusb: debug [get_api_type] matched driver name against Composite API
> [138.093750] [000009e0] libusb: debug [winusb_get_device_list] allocating new device for session [1584]
> [138.093750] [000009e0] libusb: debug [usbi_connect_device] core.usbi_connect_device
> [138.093750] [000009e0] libusb: debug [get_api_type] driver(s): TcUsb

LudovicRousseau, 2019-07-26:
Note that windows XP support should be **removed** in the next version.
See https://github.com/libusb/libusb/issues/409

mcuee, 2019-12-07:
Close this for now. Windows XP is no longer supported.

mcuee, 2020-01-07:
Close this as Windows XP is really too old. Please reopen if needed,

closed 2020-01-07 by mcuee