libusb/libusb #595: libusb_init() fails with Other error [-99] if in chroot and --disable-udev

fabled, 2019-07-16:
I am getting:
```
Failed to get USB context: failed to init libusb: Other error [-99] (g-usb-context-error-quark, 0)
```
When running in chroot where `/dev/bus/usb` is empty. Apparently no error is given if `udev` is being used.

I would like to clarify if `libusb_init` is expected to succeed even if no USB bus is present. And to have consistent behavior regardless of if using `--disable-udev` or `--enable-udev`.

For background, see: https://github.com/hughsie/fwupd/issues/1219

mcuee, 2019-07-30:
It is better to ask in the libusb-devel maining list for questions.

mcuee, 2019-12-07:
Please use libusb mailing list. Thanks.

fuzzyeric, 2021-07-13:
Why was this bug report closed?  I'm here two years later, trying to figure out why I am getting "Fu-WARNING **: FuMain: failed to get USB context: failed to init libusb: Other error [-99]" errors and there is ZERO information here about this libusb bug or its resolution.  I have non-USB devices requiring firmware updates to work correctly and libusb is a showstopper for me.  "Use the mailing list" is absolutely useless resolution information.

mcuee, 2021-07-13:
Good point. Apparently this ticket was closed wrongly. 

Whether this is an issue that is another story. 

Personally I did not think it is an issue at the time. And I still think it is not a real issue for libusb  now.  If there is no USB controller present, I think there is nothing wrong for libusb to fail libusb_init.

But I will let other libusb admins and contributors to chime in for their opinions.

mcuee, 2021-07-13:
Please also refer to past issue #511.
https://github.com/libusb/libusb/issues/511

When the container does not support libudev (--disable-udev) and Netlink, libusb_init() will fail.

mcuee, 2021-07-13:
@fuzzyeric And also I do not understand why this becomes a showstopper for you in fwupd since you are using non-USB device and that libusb_init failing is just an warning for the application. I assume it has other codes to deal with non USB devices.

mcuee, 2021-07-13:
On the other hand, there is a potential useful enhancement to libusb here. But the author has closed the pull-request. Maybe you can take a look whether it is useful info or not.
https://github.com/libusb/libusb/pull/559

hjelmn, 2021-07-14:
IMHO even if we don't document it I expect libusb_init may fail if no USB bus exists. This is not alinusb bug.

hjelmn, 2021-07-14:
I will see about updating the documentation of needed.

mcuee, 2021-07-14:
One the issue raised in https://github.com/fwupd/fwupd/issues/1219 is that libusb_init() error codes are not well documented.

mcuee, 2026-03-11:
PR #1756 has been merged. I tend to think this can be closed.

@fabled 
Please help to check if #1756 addresses the issues.
* #1756