libusb/libusb #596: Including libusb on Android using cmake

iliis, 2019-07-16:
This is one of these unfortunate cases where several things come together to cause a problem. I'm sharing it here to discuss it and get some comments and maybe spare someone else the pain of having to discover this the hard way.

## The Context

I'm porting a relatively large codebase to Android, which uses cmake to build everything and makes use of libusb to talk to some hardware. I'm cross-compiling this library by directly invoking cmake, i.e. basically `cmake .. -DANDROID_ABI=arm64-v8a -DCMAKE_TOOLCHAIN_FILE=$NDK/build/cmake/android.toolchain.cmake` (I then later include the resulting .so file in my Android project, together with some c++ glue code).

The official way to automatically find libusb is by using pkg-config. The autotools will put a `libusb-1.0.pc` file somewhere in `/usr/lib` that tells the user's build system where it can find the library and headers. We make use of that in our cmake scripts by `include(FindPkgConfig)` and `PKG_CHECK_MODULES(LIBUSB1 REQUIRED libusb-1.0>=1.0.17)`.

## The Problem

The Android build system provides no pkg-config at all, so when I build our library for Android cmake finds the host's pkg-config (or nothing at all) which then happily proceeds to include the host's libusb. Which is of course totatlly the wrong one.

## Possible Solutions

Apart from just hardcoding a few absolute paths here and there, I've come up with a few ways to fix this:

1. Write a `FindLibusb.cmake` script and include it in my project. This could fall back to pkg-config in the general case and add some magic in case we're compiling for Android. This means I have to modify our `CMakeLists.txt` to use `find_package()` instead.
2. Provide a stub that impersonates pkg-config and outputs the right values (which is what ffmpeg does to find the freetype library: https://github.com/rstiller/android-ndk-builder/blob/master/ffmpeg/fake-pkg-config)
3. Write a custom pkg-config `libusb-1.0.pc` and use environment variables to tell pkg-config where to look for it (PKG_CONFIG_{DIR,LIBDIR,SYSROOT}).

## Discussion

All these solutions are relatively simple to implement, but all require some work. I.e. it's not working out of the box. Maybe we can find a way to add support for this scenario? After all, libusb works very nicely on Android, even without root.

Maybe we can configure ndk-build to spit out a `libusb-1.0.pc` as well? I'm not familiar with ndk-build, but the newer, cmake-based Android build system would certainly be able to do this.

## P.S.

https://github.com/libusb/libusb/issues/31 bites us here, as all the pkg-config stuff uses `libusb-1.0`, whereas the Android build scripts produce `libusb1.0`

xloem, 2020-12-23:
`PKG_CONFIG_EXECUTABLE=/bin/false` https://cmake.org/cmake/help/latest/module/FindPkgConfig.html#variable:PKG_CONFIG_EXECUTABLE

iliis, 2021-01-04:
> `PKG_CONFIG_EXECUTABLE=/bin/false` [cmake.org/cmake/help/latest/module/FindPkgConfig.html#variable:PKG_CONFIG_EXECUTABLE](https://cmake.org/cmake/help/latest/module/FindPkgConfig.html#variable:PKG_CONFIG_EXECUTABLE)

You need to set `PKG_CONFIG_EXECUTABLE` to your own stub for option 2, but I don't see how replacting it with `/bin/false` would help at all?

xloem, 2021-01-04:
In a project I was working on, it was helpful to disable pkg-config when building for android.  Setting it to /bin/false makes cmake recognise that it isn't usable.  That way the user's non-android libraries wouldn't get automatically detected, and this made it easier to make reliable temporary bandaids with flags or patches.  Maybe this is not helpful for you; if so I apologise.

iliis, 2021-01-05:
Ah, so you at least get a proper error message instead of the build system silently using the completely wrong library. This is actually a neat trick, thanks :)

mcuee, 2021-02-26:
Close this for now. Please reopen if you feel the need.

iliis, 2021-03-02:
> Close this for now. Please reopen if you feel the need.

Why close it? This issue still remains unsolved.

mcuee, 2021-03-02:
To me this is out of scope for libusb project.

xloem, 2021-03-02:
Sounds like there are two separate issues here: norms for linking to this library on android build systems, and providing cmake support in libusb.  It may be possible to install pkg-config in a local prefix for android builds as a workaround.

mcuee, 2021-03-03:
Enhancements to the Android backend are welcome. Supporting CMake is out of scope. But adding pkg-config script is good.

mcuee, 2021-08-04:
@xolem Does #874 covers this one?

adisuciu, 2021-08-04:
You mispelled his name :) @xloem

mcuee, 2021-08-04:
@xloem Does #874 covers this one? Does #874 cover #31?

xloem, 2021-08-04:
@mcuee 874 and these two PRs are presently orthogonal i.e. no it does not and I expect the changes should merge fine (but haven't looked recently).

crroush, 2021-11-28:
In the UHD repo there is a [FindLIBUSB.cmake](https://github.com/EttusResearch/uhd/blob/master/host/cmake/Modules/FindLIBUSB.cmake) that *will* work with the PR #874 provided after you build you copy the libusb.h header file and libusb-1.0.so to the CMAKE_SYSROOT/include/libusb-1.0/libusb.h and CMAKE_SYSROOT/lib  

Then it will *find* the file and do the correct linking.  Because of how Android package system works, you will need to add to your gradle file 
```
sourceSets.main {
                jniLibs.srcDir = ['CMAKE_SYSROOT/libs']  
    }
```
You will need to build with:
```
ndk-build USE_PC_NAME=1
```
so it makes the properly named libusb-1.0.so vs libusb1.0.so (default)