libusb/libusb #601: libusb_get_string_descriptor_ascii always return -6 in arrived callback function

MihanLee, 2019-07-24:
I want to get USB devices on linux, but ubuntu 12.04 use old version's libusb. So I install the latest one(1.0.22). It works normally in getting descriptor by device list, but has problem in getting  descriptor by callback function that always return -6 when libusb_get_string_descriptor_ascii(). What should I do to fix it , here's my code. Thanks~
`int LIBUSB_CALL USBKeyboardCheck::USB_Arrived_Callback(libusb_context *ctx, libusb_device *dev, libusb_hotplug_event event, void *userdata)
{
    struct libusb_device_handle *handle;
    struct libusb_device_descriptor desc;
    unsigned char buf[512];
    QString exchangeBuf;
    int rc = 0;

    libusb_get_device_descriptor(dev, &desc);
    qDebug("Add usb device: \n");
    qDebug("\tCLASS(0x%x) SUBCLASS(0x%x) PROTOCOL(0x%x) idProduct(0x%x)\n",desc.bDeviceClass,
                                                                           desc.bDeviceSubClass,
                                                                           desc.bDeviceProtocol,
                                                                           desc.idProduct);
    qDebug("\tVENDOR(0x%x) PRODUCT(0x%x) iManufacturer(0x%x)\n", desc.idVendor,
                                                                 desc.idProduct,
                                                                 desc.iManufacturer);
    rc = libusb_open(dev, &handle);
    if (LIBUSB_SUCCESS != rc)
    {
        qDebug("Could not open new USB device\n");
        return 0;
    }

//    libusb_reset_device(handle);

    memset(buf, 0, sizeof(buf));
    rc = libusb_get_string_descriptor_ascii(handle, desc.iManufacturer, buf, sizeof(buf));
    if (rc < 0)
    {
        qDebug()<<"Get Manufacturer failed"<<rc;
    }
    else
    {
        qDebug("\tManufacturer: %s", buf);
    }
    exchangeBuf = QString::fromLocal8Bit((char *)buf,sizeof(buf));
    if( exchangeBuf.contains("keyboard",Qt::CaseInsensitive))
    {
        qDebug()<<"keyboard add";
        g_KeyboardDesc = desc;
    }

    memset(buf, 0, sizeof(buf));
    rc = libusb_get_string_descriptor_ascii(handle, desc.iProduct, buf, sizeof(buf));
    if (rc < 0)
    {
        qDebug()<<"Get Product failed"<<rc;
    }
    else
    {
        qDebug("\tProduct: %s", buf);
    }
    exchangeBuf = QString::fromLocal8Bit((char *)buf,sizeof(buf));
    if( exchangeBuf.contains("keyboard",Qt::CaseInsensitive))
    {
        g_KeyboardDesc = desc;
    }

    memset(buf, 0, sizeof(buf));
    rc = libusb_get_string_descriptor_ascii(handle, desc.iSerialNumber, buf, sizeof(buf));
    if (rc < 0)
    {
        qDebug("Get SerialNumber failed\n");
    }
    else
    {
        qDebug("\tSerialNumber: %s\n", buf);
    }
    exchangeBuf = QString::fromLocal8Bit((char *)buf,sizeof(buf));
    if( exchangeBuf.contains("keyboard",Qt::CaseInsensitive))
    {
        g_KeyboardDesc = desc;
    }

    libusb_close(handle);

    return 0;
}

int LIBUSB_CALL USBKeyboardCheck::USB_Left_Callback(libusb_context *ctx, libusb_device *dev, libusb_hotplug_event event, void *userdata)
{
    struct libusb_device_descriptor desc;

    libusb_get_device_descriptor(dev, &desc);

    if ( g_KeyboardDesc.iProduct == desc.iProduct && g_KeyboardDesc.iManufacturer == desc.iManufacturer)
    {
        qDebug()<<"keyboard remove\n";
    }
    qDebug("Remove usb device: CLASS(0x%x) SUBCLASS(0x%x) iSerialNumber(0x%x)\n",
            desc.bDeviceClass, desc.bDeviceSubClass, desc.iSerialNumber);
    return 0;
}

int USBKeyboardCheck::Thread()
{
    libusb_hotplug_callback_handle usb_arrived_handle;
    libusb_hotplug_callback_handle usb_left_handle;
    libusb_context *ctx;
    int rc;

    struct libusb_device_handle *handle;
    struct libusb_device_descriptor desc;
    unsigned char buf[1024];
    QString exchangeBuf;
    libusb_init(&ctx);

    libusb_device **list;
    int currentDevice = libusb_get_device_list(ctx,&list);
    if (currentDevice < 0)
        goto failure;
    for( int n = 0; n < currentDevice; n++)
    {
        libusb_device *dev = list[n];
        libusb_get_device_descriptor(dev, &desc);
        rc = libusb_open(dev, &handle);
        qDebug("No.%d:",n+1);
        qDebug("\tCLASS(0x%x) SUBCLASS(0x%x) PROTOCOL(0x%x) idProduct(0x%x)\n",desc.bDeviceClass,
                                                                               desc.bDeviceSubClass,
                                                                               desc.bDeviceProtocol,
                                                                               desc.idProduct);
        qDebug("\tVENDOR(0x%x) PRODUCT(0x%x) iManufacturer(0x%x)\n", desc.idVendor,
                                                                     desc.idProduct,
                                                                     desc.iManufacturer);
        if (LIBUSB_SUCCESS != rc)
        {
            qDebug("Could not open USB device No.%d\n",n+1);
            continue;
        }

        memset(buf, 0, sizeof(buf));
        rc = libusb_get_string_descriptor_ascii(handle, desc.iManufacturer, buf, sizeof(buf));
        if (rc < 0)
        {
            qDebug("\tGet Manufacturer failed\n");
        } else {
            qDebug("\tManufacturer: %s\n", buf);
        }
        exchangeBuf = QString::fromLocal8Bit((char *)buf,512);
        if( exchangeBuf.contains("keyboard",Qt::CaseInsensitive))
        {
            qDebug()<<"has keyboard";
            g_KeyboardDesc = desc;
            emit KeyboardIn();
        }

        memset(buf, 0, sizeof(buf));
        rc = libusb_get_string_descriptor_ascii(handle, desc.iProduct, buf, sizeof(buf));
        if (rc < 0)
        {
            qDebug("\tGet Product failed\n");
        }
        else
        {
            qDebug("\tProduct: %s\n", buf);
        }
        exchangeBuf = QString::fromLocal8Bit((char *)buf,512);
        if( exchangeBuf.contains("keyboard",Qt::CaseInsensitive))
        {
            qDebug()<<"has keyboard";
            g_KeyboardDesc = desc;
            emit KeyboardIn();
        }

        libusb_close(handle);
    }
    libusb_free_device_list(list,1);
    list = NULL;

    rc = libusb_hotplug_register_callback(ctx, LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED,
        LIBUSB_HOTPLUG_NO_FLAGS, LIBUSB_HOTPLUG_MATCH_ANY, LIBUSB_HOTPLUG_MATCH_ANY,
        LIBUSB_HOTPLUG_MATCH_ANY, USB_Arrived_Callback, NULL, &usb_arrived_handle);
    if (LIBUSB_SUCCESS != rc)
    {
        qDebug("Error to register usb arrived callback\n");
        goto failure;
    }

    rc = libusb_hotplug_register_callback(ctx, LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT,
        LIBUSB_HOTPLUG_NO_FLAGS, LIBUSB_HOTPLUG_MATCH_ANY, LIBUSB_HOTPLUG_MATCH_ANY,
        LIBUSB_HOTPLUG_MATCH_ANY, USB_Left_Callback, NULL, &usb_left_handle);
    if (LIBUSB_SUCCESS != rc)
    {
        qDebug("Error to register usb left callback\n");
        goto failure;
    }

    while (1)
    {
        libusb_handle_events_completed(ctx, NULL);
//        libusb_handle_events(ctx);
        usleep(1000);
    }

    libusb_hotplug_deregister_callback(ctx, usb_arrived_handle);
    libusb_hotplug_deregister_callback(ctx, usb_left_handle);
    libusb_exit(ctx);

    return 0;

failure:
    libusb_exit(ctx);
    return EXIT_FAILURE;
}`

mcuee, 2019-07-30:
It is better to ask in the libusb-devel maining list for questions.

MihanLee, 2019-08-05:
@mcuee Thank you. BTW, could you Tell me how to submit a new mail to mailing list. I just sent a massage to _cdickens_ just now XD.

sujx0123, 2019-12-24:
i have the same problem with you, and have you already solved it ?

mcuee, 2020-01-03:
https://github.com/libusb/libusb/wiki#Support
Please use mailing list for support.

closed 2020-01-03 by mcuee