libusb/libusb #602: SONAME change in 1.0.23 rc

jakirkham, 2019-07-25:
Appears there is [a SONAME change in the rc]( https://abi-laboratory.pro/index.php?view=objects_report&l=libusb&v1=1.0.22&v2=1.0.23-rc1 ). Curious if this is sticking around for the full release of 1.0.23 or if that is getting changed back.

LudovicRousseau, 2019-07-26:
This change was done in https://github.com/libusb/libusb/commit/fde20bb9b7cba0ea6e96db920d0a0169c361ca92

But the commit does not provide a justification.

I don’t remember if we broke the ABI with the next release.

jakirkham, 2019-07-26:
Any thoughts, @hjelmn? 🙂

flibitijibibo, 2019-07-31:
We just found this while integrating libusb into SDL, seems the only difference between 1.0.22 and master is exclusively new APIs in libusb.h, so there shouldn't be any breakage (unless I've missed something):

https://github.com/libusb/libusb/compare/v1.0.22...master

sezero, 2019-08-07:
The correct way of handling libtool versioning is incrementing lt_age
along with lt_current, because only new apis are added.  This is way the
libraries stay compatible, i.e. the main libusb library names stay the same.
Something like (against current git):
```
diff --git a/configure.ac b/configure.ac
index 76f67d9..da8a158 100644
--- a/configure.ac
+++ b/configure.ac
@@ -24,5 +24,5 @@ AC_INIT([libusb],[LIBUSB_MAJOR[.]LIBUSB_MINOR[.]LIBUSB_MICRO[]LIBUSB_RC],[libusb
 lt_current=2
 lt_revision=0
-lt_age=1
+lt_age=2
 LTLDFLAGS="-version-info ${lt_current}:${lt_revision}:${lt_age}"
 
```

With this, it compiles for Linux like:
`libusb-1.0.so -> libusb-1.0.so.0.2.0` and
`libusb-1.0.so.0 -> libusb-1.0.so.0.2.0`

.. and for Mac OS X like:
`libusb-1.0.dylib -> libusb-1.0.0.dylib`
The dylib compatibility version is incremented though (which is a true annoyance with libtool):
For libusb-1.0.22:
`/usr/local/lib/libusb-1.0.0.dylib (compatibility version 2.0.0, current version 2.0.0)`
For current git after above patch:
`/usr/local/lib/libusb-1.0.0.dylib (compatibility version 3.0.0, current version 3.0.0)`

LudovicRousseau, 2019-08-07:
I created PR https://github.com/libusb/libusb/pull/605

hjelmn, 2019-08-09:
Fixed in rc2.

closed 2019-08-09 by hjelmn