libusb/libusb #607: codump when a camera device out

benbryantzoro1, 2019-08-13:
when access a camera device in ubuntu 18.04 which use libusb, open the camera with software,when the camera was used, we unpin the device ,the libusb coredum in handle_iso_completion, we analysis the error with gdb, we can see tpriv->iso_urbs is free but the
urb belong to it callback with ioctl, do you now how to resolve it with the urb callback no in order

hjelmn, 2019-08-14:
Can you provide more information? What version are you running? What is the backtrace?

benbryantzoro1, 2019-08-15:
backtrace as follow:
#0  0x00007fc9aca89a48 in handle_iso_completion (urb=0x7fc998019480, itransfer=0x7fc97c00c740) at os/linux_usbfs.c:2358
#1  reap_for_handle (handle=0x7fc9a0002170) at os/linux_usbfs.c:2556
#2  op_handle_events (ctx=0x1e59ee0, fds=0x7fc998002000, nfds=1, num_ready=<optimized out>) at os/linux_usbfs.c:2612
#3  0x00007fc9aca8544f in handle_events (ctx=ctx@entry=0x1e59ee0, tv=tv@entry=0x7fc9a5c1ac18) at io.c:2095
#4  0x00007fc9aca85c24 in libusb_handle_events_timeout_completed (ctx=0x1e59ee0, tv=tv@entry=0x7fc9a5c1ac68, completed=completed@entry=0x7fc9a5c1ac94) at io.c:2179
#5  0x00007fc9aca85d66 in libusb_handle_events_completed (ctx=<optimized out>, completed=completed@entry=0x7fc9a5c1ac94) at io.c:2278

(gdb) p *tpriv
$2 = {{urbs = 0x0, iso_urbs = 0x0}, reap_action = NORMAL, num_urbs = 4, num_retired = 1, reap_status = LIBUSB_TRANSFER_COMPLETED, iso_packet_offset = 2}

the sefault code is here, the iso_urbs addr is 0x0,when urb == tpriv->iso_urbs[i]:
	for (i = 0; i < num_urbs; i++) {
		if (urb == tpriv->iso_urbs[i]) {
			urb_idx = i + 1;
			break;
		}
	}

I find it free in here
	/* if we're the last urb then we're done */
	if (urb_idx == num_urbs) {
		usbi_dbg("last URB in transfer --> complete!");
		free_iso_urbs(tpriv);
		usbi_mutex_unlock(&itransfer->lock);
		return usbi_handle_transfer_completion(itransfer, status);
	}

when the urb_idx no return in order, the codump happen, is this a BUG?

hjelmn, 2019-08-15:
Seems like a bug. Let me take a look.

hjelmn, 2019-08-15:
Can you configure with ```--enable-debug-log``` and run with the LIBUSB_DEBUG environment variable set to 3? Please post that to a gist so I can see the exact order of events.

benbryantzoro1, 2019-08-15:
the log as follow: 
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 2/4
 libusb: debug [reap_for_handle] urb type=0 status=0 transferred=27920
 libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 3/4
 libusb: debug [reap_for_handle] urb type=0 status=0 transferred=5664
 libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 4/4
 libusb: debug [handle_iso_completion] last URB in transfer --> complete!
 libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
 libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb6ac08aaa0 has callback 0x7fb6d6736f1c
 libusb: debug [submit_iso_transfer] need 4 32k URBs for transfer
 libusb: debug [reap_for_handle] urb type=0 status=0 transferred=23896
 libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/4
 libusb: debug [reap_for_handle] urb type=0 status=0 transferred=27920
 libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 2/4
 libusb: debug [reap_for_handle] urb type=0 status=0 transferred=25624
 libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 3/4
 libusb: debug [reap_for_handle] urb type=0 status=0 transferred=3928
 libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 4/4
 libusb: debug [handle_iso_completion] last URB in transfer --> complete!
 libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
 libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb6ac08acf0 has callback 0x7fb6d6736f1c
 libusb: debug [submit_iso_transfer] need 4 32k URBs for transfer
 libusb: debug [reap_for_handle] urb type=0 status=0 transferred=27928
 libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/4
 libusb: debug [reap_for_handle] urb type=0 status=0 transferred=28748
 libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 2/4
 libusb: debug [handle_iso_completion] low-level USB error -71
 libusb: debug [handle_iso_completion] low-level USB error -71
 libusb: debug [handle_iso_completion] low-level USB error -71
 libusb: debug [reap_for_handle] urb type=1 status=-71 transferred=0
 libusb: debug [handle_bulk_completion] handling completion status -71 of bulk urb 1/1
 libusb: debug [handle_bulk_completion] low level error -71
 libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
 libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb6ac028510 has callback 0x7fb6d6736d33
 libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 64
 libusb: debug [reap_for_handle] urb type=1 status=-108 transferred=0
 libusb: debug [handle_bulk_completion] handling completion status -108 of bulk urb 1/1
 libusb: debug [handle_bulk_completion] device removed
 libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
 libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb6ac028510 has callback 0x7fb6d6736d33
 libusb: debug [reap_for_handle] urb type=1 status=-108 transferred=0
 libusb: debug [handle_bulk_completion] handling completion status -108 of bulk urb 1/1
 libusb: debug [handle_bulk_completion] device removed
 libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
 libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb6ac028bd0 has callback 0x7fb6d6736d33
 libusb: debug [reap_for_handle] urb type=1 status=-108 transferred=0
 libusb: debug [handle_bulk_completion] handling completion status -108 of bulk urb 1/1
 libusb: debug [handle_bulk_completion] device removed
 libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
 libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb6ac028a60 has callback 0x7fb6d6736d33
 libusb: debug [reap_for_handle] urb type=1 status=-108 transferred=0
 libusb: debug [handle_bulk_completion] handling completion status -108 of bulk urb 1/1
 libusb: debug [handle_bulk_completion] device removed
 libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
 libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb6ac00b590 has callback 0x7fb6d6736d33
 libusb: debug [reap_for_handle] urb type=1 status=-108 transferred=0
 libusb: debug [handle_bulk_completion] handling completion status -108 of bulk urb 1/1
 libusb: debug [handle_bulk_completion] device removed
 libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
 libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb6ac00c240 has callback 0x7fb6d6736d33
 libusb: debug [reap_for_handle] urb type=0 status=-108 transferred=0
 libusb: debug [handle_iso_completion] handling completion status -108 of iso urb 4/4
 libusb: debug [handle_iso_completion] low-level USB error -18
 libusb: debug [handle_iso_completion] low-level USB error -18
 libusb: debug [handle_iso_completion] device removed
 libusb: debug [handle_iso_completion] last URB in transfer --> complete!
 libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
 libusb: debug [usbi_handle_transfer_completion] transfer 0x7fb6ac08acf0 has callback 0x7fb6d6736f1c
 libusb: debug [reap_for_handle] urb type=240 status=-108 transferred=0
 libusb: debug [linux_get_device_address] getting address for device: 1-8 detached: 1
 libusb: debug [udev_hotplug_event] udev hotplug event. action: remove.

if you find the way to fix this ,please tell me, Thanks!

closed 2020-01-16 by dickens

adamsunp, 2023-08-11:
Hi, i have also encountered the same problem, how to solve it ? It only appears when the link is static. I had to use a static library for some reason.

tormodvolden, 2023-08-11:
@adamsunp please provide the stack trace and debug log like the original reporter. Also, since the original report is fairly old, and your information doesn't give any reason to why it would be the same problem, please file a new bug report.

adamsunp, 2023-08-14:
> @adamsunp please provide the stack trace and debug log like the original reporter. Also, since the original report is fairly old, and your information doesn't give any reason to why it would be the same problem, please file a new bug report.

Hi, the log is as follows. Please help analyze it, thank you.  It only appears when the link is static.

libusb: debug [reap_for_handle] urb type=0 status=0 transferred=30720
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 2/4
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 30ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=0 status=0 transferred=30720
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 3/4
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 30ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=0 status=0 transferred=6144
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 4/4
libusb: debug [handle_iso_completion] last URB in transfer --> complete!
libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 5000ms
libusb: debug [usbi_handle_transfer_completion] transfer 0x55876f9358 has callback 0x7f7938a098
libusb: debug [submit_iso_transfer] need 4 32k URBs for transfer
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 30ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=0 status=0 transferred=30720
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/4
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 30ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=0 status=0 transferred=30720
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 2/4
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 30ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=0 status=0 transferred=30720
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 3/4
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 30ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=0 status=0 transferred=6144
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 4/4
libusb: debug [handle_iso_completion] last URB in transfer --> complete!
libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 5000ms
libusb: debug [usbi_handle_transfer_completion] transfer 0x55877115b8 has callback 0x7f7938a098
libusb: debug [submit_iso_transfer] need 4 32k URBs for transfer
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 30ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=0 status=0 transferred=30720
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/4
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 30ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=0 status=0 transferred=30720
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 2/4
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 30ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=0 status=0 transferred=30720
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 3/4
libusb: debug [reap_for_handle] urb type=0 status=0 transferred=6144
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 4/4
libusb: debug [handle_iso_completion] last URB in transfer --> complete!
libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 5000ms
libusb: debug [usbi_handle_transfer_completion] transfer 0x5587729818 has callback 0x7f7938a098
libusb: debug [submit_iso_transfer] need 4 32k URBs for transfer
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 30ms
libusb: debug [handle_events] poll() returned 1
libusb: debug [reap_for_handle] urb type=0 status=0 transferred=30720
libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/4
[... 1584 lines cut ...]
libusb: debug [handle_iso_completion] low-level USB error -71
libusb: debug [handle_iso_completion] low-level USB error -71
libusb: debug [handle_iso_completion] low-level USB error -71
libusb: debug [handle_iso_completion] low-level USB error -71
libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusb: debug [handle_events] poll() 4 fds with timeout in 30ms
[... cut ...]
libusb: debug [handle_iso_completion] low-level USB error -18
libusb: debug [handle_iso_completion] device removed
libusb: debug [handle_iso_completion] last URB in transfer --> complete!
libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 5000ms
libusb: debug [usbi_handle_transfer_completion] transfer 0x5587729818 has callback 0x7f7938a098
libusb: debug [libusb_cancel_transfer]
libusb: debug [libusb_cancel_transfer] cancel transfer failed error -5
libusb: debug [reap_for_handle] urb type=0 status=-108 transferred=0

adamsunp, 2023-08-14:
> @adamsunp please provide the stack trace and debug log like the original reporter. Also, since the original report is fairly old, and your information doesn't give any reason to why it would be the same problem, please file a new bug report.

Hi, the usb version I use is 1.0.19 and the ubuntu version is 18.04. Is this problem related to the version?

mcuee, 2023-08-14:
> Hi, the usb version I use is 1.0.19 and the ubuntu version is 18.04. Is this problem related to the version?

Please use libusb-1.0.26 release to see if that helps. libusb-1.0.19 is really too old.

If that does not work, try another computer with newer version of Ubuntu (say 22.04). It could be that the kernel is a bit old for your use case.

mcuee, 2023-08-14:
But it is strange that you say the issue is only happening with static linking. You may post the debug log of both cases (static linking and dynamic linking), with libusb-1.0.26. We will not support 1.0.19 now.

adamsunp, 2023-08-15:
> But it is strange that you say the issue is only happening with static linking. You may post the debug log of both cases (static linking and dynamic linking), with libusb-1.0.26. We will not support 1.0.19 now.

@mcuee Hi, i found this issue fixed in version 1.0.24, and i have been able to solve this problem with the new version. Thank you for your advice and help.

2020-12-09: v1.0.24
* Linux: Wait until all URBs have been reaped before freeing them (#607)

commit 9eda802d (2020-01-16): linux_usbfs: Wait until all URBs have been reaped before freeing them

Prior to this change, the URBs allocated for an individual transfer were
freed when the last URB in the transfer was reaped. Normally this causes
no issues because URBs are reaped in the order they were submitted. If
the device is disconnected while multiple URBs are queued, these URBs
may be reaped in an order that does not match that of submission.

Change the logic to free the URBs when all the URBs of a transfer have
been reaped rather than the last one. While in here, improve some debug
messages.

Closes #607