libusb/libusb #609: Build error on Windows

kdart, 2019-08-13:
I'm using vcpgk on Windows to build the libusb library. Since I'm having problems with getting it to work on Windows with the current release, I thought I'd try the HEAD build.

> .\vcpkg.exe install --head  libusb:x64-windows-static

But it fails to compile:

> cl : Command line error D8016 : '/source-charset:utf-8' and '/utf-8' command-line options are incompatible

kdart, 2019-08-13:
Also, trying to build natively using the DDK, as described in the INSTALL_WIN.txt, also fails.

Compiling - windows_winusb.c
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2427) : error C2143: syntax error : missing ';' before 'type'
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2428) : error C2143: syntax error : missing ';' before 'type'
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2436) : error C2065: 'handle_priv' : undeclared identifier
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2436) : error C2223: left of '->interface_handle' must point to struct/union
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2436) : error C2065: 'handle_priv' : undeclared identifier
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2436) : error C2223: left of '->interface_handle' must point to struct/union
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2437) : error C2065: 'handle_priv' : undeclared identifier
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2437) : error C2223: left of '->interface_handle' must point to struct/union
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2437) : error C2065: 'handle_priv' : undeclared identifier
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2437) : error C2223: left of '->interface_handle' must point to struct/union
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2438) : error C2065: 'priv' : undeclared identifier
1>c:\users\keith.dart\opensrc\libusb\libusb\os\windows_winusb.c(2438) : error C2223: left of '->usb_interface' must point to struct/union
Compiling - generating code...
Building Library - objfre_win7_amd64\amd64\libusb-1.0.lib
1>link : error LNK1181: cannot open input file 'c:\users\keith.dart\opensrc\libusb\libusb\os\objfre_win7_amd64\amd64\poll_windows.obj'
BUILD: Finish time: Tue Aug 13 13:07:11 2019
BUILD: Done

    16 files compiled - 6 Warnings - 13 Errors
    1 library built - 1 Error
Build failed

hjelmn, 2019-08-13:
By current release do you mean 1.0.23-rc2 or 1.0.22?

kdart, 2019-08-13:
I mean HEAD. I see that you have some Windows fixes in there that may help my other problem. I also see that your Travis CI is seeing the same compile problem. ;-)

hjelmn, 2019-08-14:
I can't fix the issue myself as I do not develop on Windows. If you find a way to fix and submit a PR I will take a look. I plan to cut a 1.0.23-rc3 tonight.

kdart, 2019-08-14:
Oh, ok. I'll try. I don't develop on Windows, either. :-) But I have to get this to run on Windows for other reasons. I'm not a Windows expert, so I was hoping one of you might have some hints on how to get this to work.

The real issue I'm having is that I can't open the device. It enumerates, I get the un-opened device, try to open it, then that fails with an error saying it doesn't exist (is not plugged in). But I know it is plugged in. I've tried all the zadig back-end combinations but none work. 

But I'll see if I can fix the Windows build problems and try it out.

hjelmn, 2019-08-14:
I took a quick look and I think I see the problem. Microsoft has historically been the worst vendor when it comes to C-language support. C89 doesn't support the code on windows_winusb.c:L2424 . All declarations must come before code in C89.

One fix is to move that code to after the variable declarations. If that works I can make the change.

hjelmn, 2019-08-14:
Yup. That was it. C99 is 20 years old now.....

mcuee, 2019-12-07:
So we can close this issue now that fix has been committed. If not, please reopen.

closed 2019-12-07 by mcuee