libusb/libusb #610: SEGV or abort when libusb_exit called during synchronous transfer

hjelmn, 2019-08-14:
I am not 100% sure this should not be considered an application error. It is relatively easy to fix so we might as well fix it and emit a warning. Documenting the issue in a bug for future reference.

If a multithreaded application does the following it could lead to a crash:

1) call libusb_init()
2) open a device and interface
3) submit a *blocking* transfer API call from thread A
4) close the device and call libusb_exit() from thread B

When the transfer times out thread A enters handle_events() which detects that the number of fds is less than required (it is 0 since the context no longer exists but we still have a reference to the device and the context through the transfer). This causes handle_events() to fail an assetion and abort().

hjelmn, 2019-08-14:
@mcuee @LudovicRousseau What do you think. The simple fix is:

```patch
--- libusb/io.c
+++ libusb/io.c
@@ -2094,6 +2094,13 @@
         * libusb_handle_events() from within a hotplug or transfer callback) */
        usbi_mutex_lock(&ctx->event_data_lock);
        r = 0;
+       if (0 == ctx->pollfds_cnt) {
+               /* context has been torn down. the error will be
+                * returned in the transfer */
+               usbi_mutex_unlock(&ctx->event_data_lock);
+               return LIBUSB_SUCCESS;
+       }
+
```

hjelmn, 2019-08-14:
Actually, that fix is too simplistic. It is working in my case but that is dumb luck. Will see if there is some other simple fix that can at least avoid a crash.

hjelmn, 2019-08-14:
Pushed a slightly better fix. There is still a chance of accessing an invalid context but the window is smaller now.

mcuee, 2021-06-05:
I will close this and please refer to $703 in the future.

closed 2021-06-05 by mcuee

mcuee, 2023-02-06:
1) https://github.com/libusb/libusb/issues/703
2) https://github.com/libusb/libusb/issues/182
3) https://github.com/libusb/libusb/issues/297
4) https://github.com/libusb/libusb/issues/540
5) https://github.com/libusb/libusb/issues/610