libusb/libusb #613: Win10 + docker-for-win + virtualhere: error [_get_usbfs_fd]

1138-4EB, 2019-08-17:
I'm trying to program an Icestick connected to a Windows 10 host from a GNU/Linux container running on the same host (docker-for-win).

- First, I built and loaded the usbip kernel module for the underlying Alpine Linux VM.
- Then, I started a VirtualHere server on the host.
- Last, I started a VirtualHere client inside a docker container and told it to use the device.

From the container, I can see the device with lsusb:

``` bash
root@781eb7002579:/# lsusb
Bus 001 Device 058: ID 0403:6010
Bus 001 Device 001: ID 1d6b:0002
```

I can find it with ftdi's `find_all`, but it fails at opening it:

``` bash
root@781eb7002579:/# find_all
Number of FTDI devices found: 1
Checking device: 0
ftdi_usb_get_strings failed: -4 (libusb_open() failed)
```

With libusb's `xusb`, it cannot find the device:

```bash
root@781eb7002579:/# xusb -i 0403:6010
Using libusb v1.0.22.11312

Opening device 0403:6010...
libusb: error [_get_usbfs_fd] File doesn't exist, wait 10 ms and try again
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/058: No such file or directory
  Failed.
```

virtualhere, 2019-08-18:
Im wondering if it might be asleep, could you turn off usb suspend on the host https://www.windowscentral.com/how-prevent-windows-10-turning-usb-devices

1138-4EB, 2019-08-18:
> Im wondering if it might be asleep, could you turn off usb suspend on the host [windowscentral.com/how-prevent-windows-10-turning-usb-devices](https://www.windowscentral.com/how-prevent-windows-10-turning-usb-devices)

Thanks for reaching out! I tried turning it off as you suggested, but unfortunately it makes no difference. This is the specific procedure I am using: https://github.com/ghdl/docker/tree/usbip/usbip.

Regarding the first comment above, the error produced by libusb is because `/dev/bus/usb/001/058` does not exist. This is true. `/dev/bus/usb/001/` does exist, and it is a directory, but `058` does not. See:

```bash
# ./run.sh -e ls -la //dev/bus/usb/
> /usr/bin/winpty docker exec -it vhclient ls -la //dev/bus/usb/
total 0
drwxr-xr-x 3 root root 60 Aug 18 18:20 .
drwxr-xr-x 3 root root 60 Aug 18 18:20 ..
drwxr-xr-x 2 root root 60 Aug 18 18:20 001

# ./run.sh -e ls -la //dev/bus/usb/001/
> /usr/bin/winpty docker exec -it vhclient ls -la //dev/bus/usb/001/
total 0
drwxr-xr-x 2 root root     60 Aug 18 18:20 .
drwxr-xr-x 3 root root     60 Aug 18 18:20 ..
crw-rw---- 1 root root 189, 0 Aug 18 18:20 001

# ./run.sh -e ls -la //dev/bus/usb/001/001
> /usr/bin/winpty docker exec -it vhclient ls -la //dev/bus/usb/001/001
crw-rw---- 1 root root 189, 0 Aug 18 18:20 //dev/bus/usb/001/001

# ./run.sh -e file //dev/bus/usb/001/001
> /usr/bin/winpty docker exec -it vhclient file //dev/bus/usb/001/001
//dev/bus/usb/001/001: character special (189/0)
```

I don't know if this is an issue with VirtualHere, or if libusb should be able to manage the device regarless of `/dev/bus/usb/001/*` not being valid.

virtualhere, 2019-08-18:
VirtualHere doesnt create the device file, its created by the kernel. Is it possible to get the syslog of the docker image and see if there is an error in there?

1138-4EB, 2019-08-19:
> Is it possible to get the syslog of the docker image and see if there is an error in there?

I checked `/var/log`, which is empty. Regarding `dmesg`:

```
insmod usbip-core.ko
[  123.545493] usbip_core: loading out-of-tree module taints kernel.
[  123.549912] usbip_core: USB/IP Core v1.0.0

insmod vhci-hcd.ko
[  162.401736] vhci_hcd vhci_hcd: USB/IP Virtual Host Controller
[  162.405562] vhci_hcd vhci_hcd: new USB bus registered, assigned bus number 1
[  162.410055] vhci_hcd: created sysfs vhci_hcd
[  162.413373] hub 1-0:1.0: USB hub found
[  162.415591] hub 1-0:1.0: 8 ports detected
[  162.418546] vhci_hcd: USB/IP 'Virtual' Host Controller (VHCI) Driver v1.0.0
```

Do you know where should I look for more detailed info?

virtualhere, 2019-08-19:
Use the device and then have a look in syslog, also you must use a
purchased virtualhere server.

1138-4EB, 2019-08-19:
Eureka! The workaround is quite strange, but I found how to make it work :D.

It turns out that, although the file which should be created by the kernel when the user requests the virtualhere client to `USE` a device is not visible to the container that created it, it is properly created and it can be seen by any other privileged container.

In terms of https://github.com/ghdl/docker/tree/usbip/usbip#usage:

```bash
# ./run.sh -l
# ./run.sh -s ghdl/synth:vhcli
> docker rm -f vhclient
Error: No such container: vhclient
> docker run -d --privileged --rm --name vhclient ghdl/synth:vhcli vhclientx86_64
04537c9d307f539c0f0147d2550f6334b71e8f4b5db03c304589f89a9d3a09d7

# ./run.sh -e lsusb
> /usr/bin/winpty docker exec -it vhclient lsusb
Bus 001 Device 001: ID 1d6b:0002

# ./run.sh -c "LIST"
> /usr/bin/winpty docker exec -it vhclient vhclientx86_64 -t LIST
VirtualHere Client IPC, below are the available devices:
(Value in brackets = address, * = Auto-Use)

Auto-Find currently on
Auto-Use All currently off
Reverse Lookup currently off
Reverse SSL Lookup currently off
VirtualHere Client not running as a service

# ./run.sh -c "MANUAL HUB ADD,10.0.75.1:7575"
> /usr/bin/winpty docker exec -it vhclient vhclientx86_64 -t MANUAL HUB ADD,10.0.75.1:7575
OK

# ./run.sh -c "LIST"
> /usr/bin/winpty docker exec -it vhclient vhclientx86_64 -t LIST
VirtualHere Client IPC, below are the available devices:
(Value in brackets = address, * = Auto-Use)

MYSRV (DESKTOP-E1ER43H:7575)
   --> Mouse (DESKTOP-E1ER43H.3)
   --> 0x1829 (DESKTOP-E1ER43H.2)
   --> icestick (DESKTOP-E1ER43H.1)

Auto-Find currently on
Auto-Use All currently off
Reverse Lookup currently off
Reverse SSL Lookup currently off
VirtualHere Client not running as a service

# ./run.sh -c "USE,DESKTOP-E1ER43H.1"
> /usr/bin/winpty docker exec -it vhclient vhclientx86_64 -t USE,DESKTOP-E1ER43H.1
OK

# ./run.sh -e lsusb
> /usr/bin/winpty docker exec -it vhclient lsusb
Bus 001 Device 004: ID 0403:6010
Bus 001 Device 001: ID 1d6b:0002

# ./run.sh -e ls -la //dev/bus/usb/001
> /usr/bin/winpty docker exec -it vhclient ls -la //dev/bus/usb/001
total 0
drwxr-xr-x 2 root root     60 Aug 19 01:12 .
drwxr-xr-x 3 root root     60 Aug 19 01:12 ..
crw-rw---- 1 root root 189, 0 Aug 19 01:12 001

# ./run.sh -e iceprog -t
> /usr/bin/winpty docker exec -it vhclient iceprog -t
init..
Can't find iCE FTDI USB device (vendor_id 0x0403, device_id 0x6010 or 0x6014).
ABORT.

# docker run --rm --privileged busybox ls -la //dev/bus/usb/001
total 0
drwxr-xr-x    2 root     root            80 Aug 19 01:16 .
drwxr-xr-x    3 root     root            60 Aug 19 01:16 ..
crw-rw----    1 root     root      189,   0 Aug 19 01:16 001
crw-rw----    1 root     root      189,   3 Aug 19 01:16 004

# docker run --rm --privileged ghdl/synth:vhclient ls -la //dev/bus/usb/001
total 0
drwxr-xr-x 2 root root     80 Aug 19 01:16 .
drwxr-xr-x 3 root root     60 Aug 19 01:16 ..
crw-rw---- 1 root root 189, 0 Aug 19 01:16 001
crw-rw---- 1 root root 189, 3 Aug 19 01:16 004

# docker run --rm --privileged ghdl/synth:vhclient iceprog -t
init..
cdone: high
reset..
cdone: low
flash ID: 0x20 0xBA 0x16 0x10 0x00 0x00 0x23 0x51 0x73 0x10 0x25 0x00 0x57 0x00 0x19 0x07 0x06 0x15 0xA5 0x19
cdone: high
Bye.
```

I'm closing this issue because it seems to be unrelated to libusb, libftdi and/or virtualhere.

closed 2019-08-19 by 1138-4EB