libusb/libusb #616: Access Violation issue on Windows

Zaaphod, 2019-08-23:
I have a device that only answers an libusb_interrupt_transfer read request when it has some data to send.  I have a thread that just constantly issues libusb_interrupt_transfer read request, and whenever it gets a response, it puts the data in some variables and loops back and starts another read request.  So most of the time this thread is doing nothing but waiting for some data to come in.

the problem I am having is if I try to do a libusb_control_transfer in another thread while the libusb_interupt_transfer is sitting around awaiting for a response, I am crashing with an Access Violation...
 either the libusb_control_transfer will cause this..  like this: 
[https://pastebin.com/KY6X3xmR](https://pastebin.com/KY6X3xmR) 
or the libusb_interrupt_transfer waiting to read will crash like this:
[https://pastebin.com/VTqFyAiB](https://pastebin.com/VTqFyAiB)

Someone who has been helping me, but only has Linux has done this same exact thing and has no issues with it at all.   He just keeps trying to read in a tight read loop and writes when he feels like it in another thread.  

Am I doing something fundamentally wrong?  or is this an issue with windows?  is there a better way I can wait for data if I have a device that will not answer a read request until it has something to send, but I want to write to it while it's waiting?  Is there some reason this method of sending data while waiting for a read in another thread works fine on Linux but not in Winodws?

My project is here:
[https://github.com/Zaaphod/libusbxhid/blob/Test/WHB04B-4_test.pas](https://github.com/Zaaphod/libusbxhid/blob/Test/WHB04B-4_test.pas)
I am running on Windows 10 professional 64bit.

Any advice is greatly appreciated.

mcuee, 2019-12-09:
libusb: error [winusbx_clear_halt] unable to match endpoint to an open interface - cannot clear
clear halt failed

The above rings a bell. Maybe the device goes bad with the clear halt feature request.

You may want to post the full debug log to see if they can provide more clues.

dickens, 2020-01-17:
Can you provide a debug log?

Zaaphod, 2020-01-17:
I don't know how to create a debug log.  Any advice on how to do that?

mcuee, 2020-01-20:
SET LIBUSB_DEBUG=4 and then run your script.

dickens, 2020-08-18:
Closing this as there has been no activity for over 6 months.  Feel free to reopen if you can provide additional information.

closed 2020-08-18 by dickens