libusb/libusb #618: Misc: To release 1.0.24

hjelmn, 2019-08-28:
Target to add windows hotplug support.

mcuee, 2020-09-01:
Maybe we can delay the Windows hotplug to 1.0.25 and have a release 1.0.24 as early as possible.

mcuee, 2020-09-01:
As per @mvp libusb 1.0.23 is broken in macOS Catalina. It also has major regression in Android.

mcuee, 2020-09-01:
https://github.com/libusb/libusb/pull/760 is good to have in 1.0.24 if possible.

mvp, 2020-09-19:
@mcuee, #760 is merged now. Can you start release process for 1.0.24 now?

mcuee, 2020-09-23:
I agree. Need to get agreement with other admins.
@hjelmn @dickens

mvp, 2020-10-20:
Any update on the 1.0.24 release?

dickens, 2020-10-21:
I have one last set of changes to get in for Windows (relating to examples) and then I can start the release.

mvp, 2020-11-08:
Hate to be that guy, but is there any update on releasing 1.0.24?

dickens, 2020-11-10:
> Hate to be that guy, but is there any update on releasing 1.0.24?

Yes, I'm finishing my last changes to the examples.  I expect to be done this week.

dickens, 2020-11-28:
I've officially started the 1.0.24 release with rc1 available [here](https://github.com/libusb/libusb/releases/tag/v1.0.24-rc1).  I'll send out the announcement to the libusb-devel mailing list shortly.

dickens, 2020-12-10:
Primary release work is finished with c6a35c5.  I'll send the announcement and upload Windows binaries to follow.

closed 2020-12-10 by dickens