libusb/libusb #619: Windows + Composite device + KBD interface -> libusb_open fails, can't claim any interfaces

neoto, 2019-08-30:
I have an issue with lib_usb failing to open the composite device on Windows if one of the interfaces is HID keyboard, automatically claimed by the system HID driver.
The issue is somehow system-dependent, since calling CreateFileA without generic R/W on the KBD interface succeeds on most systems.

I have a device that has these interfaces:
- [0] HID Joystick
- [1] HID (generic)
- [2] HID Keyboard
- [3] Bulk endpoint (WinUSB driver installed)

hid_open in windows_winusb.c tries to open file handles to each interface and if it fails in at least one (in my case the interface 2), the libusb_open returns the error.
Commenting the lines 3280 to 3290 results in perfectly functioning library for my device - interface 2 is claimed by the HID driver and I can claim interfaces 1 and 3 in my own application.
I am not sure whether such behavior is really necessary - the error should be thrown not before the 'locked' interface is accesssed.

neoto, 2019-08-30:
By writing the question I started looking for information in a different pattern and came across this thread, which explains the background: https://stackoverflow.com/questions/53761417/createfile-over-usb-hid-device-fails-with-access-denied-5-since-windows-10-180

Unfortunately, the issue is still present on a fully updated Windows 10 system.

trollcop, 2019-10-01:
Here to #metoo this thing.
Since I don't care about HID device at all, I simply removed the attempt of opening HID device here:
https://github.com/libusb/libusb/blob/master/libusb/os/windows_winusb.c#L4121
by #if 0'ing that entire bit of code.

However, I think the proper solution is to add a flag to context which ignores HID devices. or properly fix the situation with opening.

Despite what the comment says at https://github.com/libusb/libusb/blob/master/libusb/os/windows_winusb.c#L3587
what actually happens is 1st CreateFile returns FILE_NOT_FOUND (errno=2) for whatever reasons (not access denied), the 2nd CreateFile does the same, at which point hid_open() fails with LIBUSB_ERROR_NO_DEVICE and bla bla blah.

For the time being I "fixed" it by #if 0'ing the call to hid_open() as I have no use for the HID interface of my composite device anyway.

However a proper solution should probably be made. My vote goes for a flag in context somewhere which tells it to ignore HID.

mcuee, 2019-11-25:
Use HIDAPI for your generic interface. libusb's HID support is limited and unlikely to be improved any time soon now that there is no active Windows developer.

mcuee, 2020-01-07:
Close this for now. 
To OP: please reopen if you have new info.

trollcop, 2020-01-07:
???
Why is this closed, the problem is most definitely NOT solved on windows.

gregsskyles, 2020-01-07:
I think the comment in November was pretty clear: until someone steps up to be the Windows maintainer for libusb, this issue (and any other Windows-specific issues) are not going to be solved.

trollcop, 2020-01-07:
> I think the comment in November was pretty clear: until someone steps up to be the Windows maintainer for libusb, this issue (and any other Windows-specific issues) are not going to be solved.

Absolutely, but that doesn't mean the bug goes away. In fact it is still a problem, and I have to maintain a local "fixed" version of libUSB which doesn't try attaching to HID devices by default.

I've suggested some approaches how to fix it in https://github.com/libusb/libusb/issues/619#issuecomment-536954376 are you saying there's *nobody* that can look at the issue on Windows and submit a patch?

hjelmn, 2020-01-07:
IMHO. It might be time to delete the Windows HID backend. It was controversial when it was added and HID users should be using hidapi anyway. The problem is we do not know how many code rely on the Windows HID support.

mcuee, 2020-01-08:
I can reopen the issue now.

mcuee, 2020-01-08:
@hjelmn Yes now I agree to remove the HID backend from Windows, since we recommend HIDAPI for HID device. Moreover HIDAPI github repo is also under the libusb organization.

pbatard, 2020-01-08:
Well, my opinion has always been the following:

Libusb is meant to be a generic library, so, in the absolute, it should provide agnostic access to as many devices as it can. Which means that if there's a somewhat generic native API that provides access to HID devices, or hubs, or whatever else, then, even if that access is limited, libusb should be designed to provide access to those devices. Now, I understand that some may be tempted to say _"but it's too limited"_, to which I would answer that it doesn't matter, because we do have limits with all the backends anyway (WinUSB does limit how generic an access we can provide too for instance), and, again, the design goal of the library should always to provide as much access to as many devices as we can.

Which brings us to the second part, the _"as we can"_. Obviously, if we are short of manpower and trying to achieve the goal above is going to add heavily to our workload, it's pretty obvious that we may want to be flexible with the established ideal above, be it by rejecting to add support for specific OSes (eg. older versions of Windows, or niche OSes), specific backends, or removing elements that we estimate are going to become a liability moving on.

Now, when I was maintainer of the Windows backend, and despite the push in the opposite direction from a specific maintainer, I didn't see much of an issue trying to support HID in the Windows backend, because I __did__ have the time to address HID backend issues that were thrown at us, as part of the global Windows backend maintainership. And that is why I did push to keep HID in, because, technically, as long as you keep users aware of the limitations, there's no real problem with having HID in libusb.

But of course, with the lack of an active Windows backend maintainer, the situation is a bit different now, so it's really up to the active maintainership to decide what they want to do.

I'll just note that it may be worth estimating if the time and effort that may be spent removing the HID backend might not be greater than the time and effort spent simply fixing this and other HID issues as, if the reason we're removing the HID backend is _"so that we have to waste less time on it"_, and its removal results in current maintainers having to spend more time doing so than fixing the issue reported above, it does seem a bit quite counter productive to me.

Obviously, if you're going to remove HID, then you're going to have to test, on Windows, that you haven't broken anything (or at least I hope that someone here, which won't be me, will do so), which can't help to keep me a bit puzzled as to why testing a fix for the issue above is seen so drastically difficult.

Maybe time and effort would be more effectively spent trying to find an active Windows backend maintainer than starting to hack at the Windows backend and removing parts that we don't want to have to maintain. Considering the lack of volunteers for maintaining the Windows backend, I have to wonder how long it's going to be, before we decide that supporting libusb0 or libusbK is no longer worth it, because we have issues that nobody has volunteered to fix accumulating here too. Of course, it would also help if the people who report those issues, and expect the libusb team to fix it, realized that there currently isn't any active Windows maintainer, and that, if they do want something fixed, they __really__ need to invest the time submitting Pull Request, rather than leave hints as to what __somebody else__ should do.

So, to reiterate what I said above, and since I think I've made the points I wanted to make clear, I'll leave it entirely to the current maintainership to decide if they want to remove HID, noting that I'm not going to oppose that move if they decide to do so. But I'll also remind them that Open Source activity is often borne out of frustration which means that if libusb Windows users are getting frustrated that their backend is not properly being maintained, someone is bound to take matters in their own hand and step up to become its new maintainer, __eventually__...

mcuee, 2020-01-08:
Thanks Pete for the comments. I actually think in the end it is just better to remove lbusb0.sys and libusbk.sys support as well once Windows 7 is out of the picture. Also remove the dependency of libusbk.dll. 

But you are right in the end the problem is the lack of man power for Windows backend.

Anyway, the priority is probably not fixing the HID backend or lbusb0.sys unless someone provides nice fixes, rather hotplug is the real missing feature and rather complicated matter.

mcuee, 2020-01-08:
Of the three projects under Travis Robinson and I was helping him on, libusbk is in a limbo state.

libusb-win32 is in a bug fixing maintenance only mode and luckily we got a new maintainer for libusb-win32 who is able to push fixes and help in the mailing list.

libusbk is kind of in a limbo now that I am not successful in getting someone to help out yet.

Luckily libusbdotnet has a new maintainer now.

mcuee, 2020-01-09:
Of course Open Source project does have a good advantage, if someone feels the need and got some time, they can fill the voids. 

HIDAPI is a good example and now it is active again after Alan kind of disappears from the project.

pyusb was in a limbo state for a year plus and now there is a new maintainer as well. I am helping there as well.

mcuee, 2020-07-29:
Please also take note improving the Windows HID backend is not the focus of the project. In fact, we may want to remove the support of HID altogether and refer the user to HIDAPI.

If libusb built-in HID support does not work for you, please try HIDAPI.

If you want to help to improve the current Windows HID support, that will also be welcome.

trollcop, 2020-07-29:
I solved the problem in my project by going directly to WinUSB and bypassing libusb entirely. Since on windows it's just a (buggy) wrapper around WinUSB anyway.

mcuee, 2020-07-29:
> I solved the problem in my project by going directly to WinUSB and bypassing libusb entirely. Since on windows it's just a (buggy) wrapper around WinUSB anyway.

We are talking about HID here. 

But anyway, if you think the WinUSB backend is buggy and you are willing to help, you are welcome. If not, never mind.

trollcop, 2020-07-29:
> We are talking about HID here.

I know, that was the original bug. HID prevented using composite devices with libusb on Windows. I hacked it in my local copy to remove HID check but since nobody seemed to want to fix it upstream, and I didn't feel like maintaining a separate build of libusb just for that, i switched directly to WinUSB API.

the buggy part referred to device discovery and bug with hid on composite devices.

mcuee, 2020-07-29:
I see. Thanks for the clarifications.

mcuee, 2020-07-30:
https://github.com/Arakula/libusb/commit/e371e0edf84d72c35881868c27fccc84f361b445

Temporary fix by Arakula.

mcuee, 2023-01-23:
@trollcop 

Just wonder if you can try the latest git HEAD to see if the issue is still there or not. Thanks.

trollcop, 2023-01-24:
> @trollcop
> 
> Just wonder if you can try the latest git HEAD to see if the issue is still there or not. Thanks.

Hi,

I can confirm it works with the current HEAD.

I get the following in my code:

```
libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without
drv_libusb_init(): libusb_open_device_with_vid_pid(devh) = 00A5AFE0
drv_libusb_init(): libusb init ok
```

I'm guessing the HID device failure is the new code.

mcuee, 2023-01-24:
The warning message is expected for Keyboard/Mouse.

mcuee, 2023-01-24:
In this case I will close this issue.

closed 2023-01-24 by mcuee