libusb/libusb #620: Solaris backend is not correctly setting the one transfer mode for Interrupt-In pipe

kentsuji, 2019-08-31:
In sunos_check_device_and_status_open() function of os/sunos_usb.c, if the target is an interrupt In pipe, it opens the status endpoint first and writes USB_EP_INTR_ONE_XFER control to the status endpoint to make it the one transfer mode.  And it then closes the status endpoint.  After that, it opens the xfer endpoint and re-opens the status endpoint.  This code is not correct. Closing the status endpoint implicitly disables the one transfer mode.  As a result, the established interrupt-In pipe will not be working in the one transfer mode.  So, an event notification won't be delivered to the caller in timely manner.  According to the comment in the source, closing the status endpoint first and opening the xfer endpoint and status endpoint in this order is intentional to avoid an issue that may happen if the USB device has multiple configurations.   I think the issue regarding the multiple configurations is due to a ugen driver bug.  But closing the status endpoint after writing USB_EP_INTR_ONE_XFER introduced the problem I'm reporting.  Actually, there is a workaround for the multiple configurations issue without breaking the one transfer mode.  It can open the xfer endpoint first and immediately close it.  Then, it can open the status endpoint, write the USB_EP_INTR_ONE_XFER control to the status endpoint, and re-open the xfer endpoint.  With this scheme, the one transfer mode can be kept for the interrupt-In pipe without causing an issue for the USB device having multiple configurations.
Please see my sample fix for this.
[sunos_usb.c.one_transfer_mode.patch.txt](https://github.com/libusb/libusb/files/3561968/sunos_usb.c.one_transfer_mode.patch.txt)

Mno-hime, 2019-09-27:
Fixed with https://github.com/libusb/libusb/pull/629.

dickens, 2020-01-13:
Looks like this has been fixed, so closing.

closed 2020-01-13 by dickens