libusb/libusb #621: Transfer Cancelled Error With Bulk Transfer When Reading Data

napperley, 2019-09-05:
Every time a Bulk Transfer is done for reading data a transfer cancelled error occurs:
`libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=3`

This is very strange since all known instances where a transfer cancelled error occurs is with using the [Asynchronous Device IO API](http://libusb.sourceforge.net/api-1.0/group__libusb__asyncio.html). Nothing in the Lib USB documentation covers transfer cancelled errors when using the [Synchronous Device IO API](http://libusb.sourceforge.net/api-1.0/group__libusb__syncio.html), which suggests the error is unexpected. Below is the code used to read data from a USB device (uses the **CP2102 USB to UART Bridge Controller**):
```kotlin
fun runReadLoop(devHandle: CPointer<libusb_device_handle>?) = memScoped {
    val dataLength = 1500
    val data = allocArray<UByteVar>(dataLength)
    val endpoint = 0x81.toUByte()
    println("Running read loop...")
    // Ensure the device is ready for reading.
    libusb_reset_device(devHandle)
    while (true) {
        val readStatus = libusb_bulk_transfer(
                dev_handle = devHandle,
                endpoint = endpoint,
                timeout = 9000,
                length = dataLength,
                data = data,
                actual_length = null
        )
        processRead(data = data, dataLength = dataLength, readStatus = readStatus)
        sleep(1)
    }
}

fun processRead(data: CArrayPointer<UByteVar>, dataLength: Int, readStatus: Int) {
    val success = 0
    if (readStatus == LIBUSB_ERROR_IO) {
        fprintf(stderr, "Cannot read USB device.\n")
    } else if (readStatus == success) {
        val str = data.createString(dataLength)
        println("Received Data:\n$str")
    }
}
```

Here is the information for the device (using the **lsusb** command):
```
Bus 001 Device 008: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x10c4 Cygnal Integrated Products, Inc.
  idProduct          0xea60 CP210x UART Bridge / myAVR mySmartUSB light
  bcdDevice            1.00
  iManufacturer           1 Silicon Labs
  iProduct                2 CP2102 USB to UART Bridge Controller
  iSerial                 3 0001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 CP2102 USB to UART Bridge Controller
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

Running the program on Linux Mint 19.1 (64 bit) using the Lib USB library (via the **libusb-dev** package **v2:1.0.21-2**)

napperley, 2019-09-05:
The error that was encountered highlights a issue with the way that Lib USB handles errors. Lib USB doesn't mention what the error is, which means a Software Developer has to guess what the error is. Currently I am guessing that the second error (**3**) is [ENOENT](https://www.kernel.org/doc/html/latest/driver-api/usb/error-codes.html) - *Interface or endpoint doesn't exist, or isn't enabled*.

hjelmn, 2019-09-05:
The error looks like it comes from below libusb. Not much we can do with it either except translate it into our error space.

Have you checked the return code of ```libusb_reset_device```? If it returns an error your may have to reopen the device. See the documentation for the function.

napperley, 2019-09-05:
Added a check to `libusb_reset_device` and found that 0 is returned, which means that the USB device is still open. Would need a way to confirm if the USB Endpoint is enabled, provided it is possible to do so. Could be that the USB Endpoint is disabled which would explain why the reads fail.

Does Lib USB provide anything in the API to check if a USB endpoint is enabled or not? If the USB Endpoint for the USB device is disabled then there would need to be a way to enable it. Is it possible to enable a USB Endpoint with Lib USB?

mcuee, 2020-01-07:
If an endpoint does not exist, then libusb can not help much. Close this for now.

dickens, 2020-01-13:
The error shown (errno=3) is the low-level error that the Linux kernel returned.  This is the ESRCH error code, which as far as I can tell is only returned by the kernel when I/O is attempted on a device that does not have an active configuration (in other words is in configuration 0).  This should have ultimately been translated to your application as LIBUSB_ERROR_IO, which is completely appropriate.

There are no direct checks for whether an endpoint is enabled, however if you have verified that the device's configuration is set appropriately, you have claimed the interface of interest and you have set the correct alternate setting (if there are more than 1), then the endpoint is "enabled" from the host's perspective.

Going to close this for now as I don't see an issue with libusb.

closed 2020-01-13 by dickens