libusb/libusb #622: Windows 10 incompatibility for libusb-win32 1.2.6.0 binary with Code integrity checks

kartrophy, 2019-09-16:
Hi,

We are currently using the latest version libusb-win32 1.2.6.0 binary (from https://sourceforge.net/projects/libusb-win32/files/libusb-win32-releases/).

During our trial to be compatible with HCVI mode in windows 10 version 1809 or later, we run the following command verifier.exe /flags 0x02000000 /driver <driver.sys> (https://docs.microsoft.com/en-us/windows-hardware/test/hlk/testref/driver-compatibility-with-device-guard) .
When the device using the driver libusb0.sys is plugged, it causes a bluescreen with DRIVER_VERIFIED_DETECTED_VIOLATION. 

Can you please help us to solve this issue and to be complaint with windows 10 HCVI mode?

Thanks in advance for your help.

Karthiga.

kartrophy, 2019-09-16:
Also tested with LibusbK.sys and ended with same bluescreen with DRIVER_VERIFIED_DETECTED_VIOLATION.

diabolo38, 2019-09-16:
strange  i had no issue with libusbk driver (generated by libusbk inf wizard or manualy )  on many win10 system 
maybe you can try installing with zadig (libusb-win32 or libubsuk ) see if it does the same ?

anyway that is not a libusb issue but a driver install porblem.

kartrophy, 2019-09-16:
i install libusbK through zadig. Did you try to load it after activating this driver in the driver verifier with code integrity verfication? In normal use, i have no issue. 

the issue is located in the libusb0.sys and lisusbK.sys.

diabolo38, 2019-09-16:
no never did it , i do "simple" install and use ;)

kartrophy, 2019-09-25:
libusb-win32 developer community fixed it in the following release https://sourceforge.net/projects/libusb-win32/files/libusb-win32-snapshots/20190918/ and source code in git : https://github.com/mcuee/libusb-win32/commit/9afd01fc136f1f91d94a194889886b9ebb47be1c

Thanks a lot for this prompt support and release!

closed 2019-09-25 by kartrophy