libusb/libusb #623: Dose 1.0.23 work with Android?

TommySato, 2019-09-18:
I replaced libusb1.0.so from 1.0.22 to 1.0.23 on my Android studio.
But libusb_get_device_list() always reports no device.
1.0.22 is OK.
linux_default_scan_devices() is not called because of compiler switch.
I think UDEV option have to be ON when compiling. 
Do you know how to turn it on?
I compile libusb1.0.so using ndk-build command on Windows10.

laran002, 2019-11-20:
Is this a duplicate of #584

mcuee, 2020-01-06:
2f3bc98

Maybe this commit has some problems.

mcuee, 2020-01-06:
I have labeled this as duplicate for #584

mcuee, 2020-08-03:
https://github.com/libusb/libusb/pull/760

Not so sure if the above will help.

mcuee, 2020-08-26:
https://github.com/libusb/libusb/pull/760

Maybe users can try Pull Request 760.

mcuee, 2020-12-19:
Close this for now. Please go to the main thread. #584

closed 2020-12-19 by mcuee