libusb/libusb #626: Cannot obtain bus or port number

Chryseus, 2019-09-23:
These two function return no value:
libusb_get_bus_number(libusb_device*)
libusb_get_port_number(libusb_device*)

**Sample code**
```
libusb_device** DeviceList;
deviceCount = libusb_get_device_list(context, &DeviceList);
for (auto i = 0; i < deviceCount; i++)
{
auto Temp = DeviceList[i];
uint8_t bus = libusb_get_bus_number(Temp);
uint8_t port =  libusb_get_port_number(Temp);
}
```

**Test System**
Arch Linux 5.2.14-arch2-1
gcc 9.1.0
libusb 1.0.22

LudovicRousseau, 2019-09-23:
They return no value or they return 0?

From http://libusb.sourceforge.net/api-1.0/group__libusb__dev.html#ga14879a0ea7daccdcddb68852d86c00c4
```
Returns
    the port number (0 if not available) 
```

Chryseus, 2019-09-23:
> They return no value or they return 0?
> 
> From http://libusb.sourceforge.net/api-1.0/group__libusb__dev.html#ga14879a0ea7daccdcddb68852d86c00c4
> 
> ```
> Returns
>     the port number (0 if not available) 
> ```

No value / null.
I tried it after libusb_open as well but nothing.

LudovicRousseau, 2019-09-24:
What is the output of the libusb example code `./examples/listdevs`?

Chryseus, 2019-09-24:
> What is the output of the libusb example code `./examples/listdevs`?

I've solved the problem, I was using cout and forgot to typecast uint8_t to an integer.

closed 2019-09-24 by Chryseus