libusb/libusb #633: sanitize_path function for windows incorrectly works

lo1ol, 2019-10-05:
Hello. My program uses your library (v1.0.18). 

Recently, I've noticed that a lot of users have problems with it. Some of usb devices don't get recognized by the program on some computers. When I started looking for a reason of errors I've found out that for devices which contain '//' symbols, a backslash is replaced by a sharp. It happens in sanitize_path function. For example, device path "\\\\?\\hid#vid_0a89&pid_0091#7&8ad5edf&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd" was format as "\\\\.\\HID#VID_0A89&PID_0091#7&8ad5edf&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}#KBD" (last backslash was replaced by sharp).

I didn't find any reason for this behaviour and made a patch for my version of library. Everything works after this patch.

You can find it  here (https://github.com/lo1ol/libusb/tree/sharp_path_fixing in last commit).

Can you say if my patch is right? And if not, can you please explain the reason for this replacement.

mcuee, 2019-12-10:
Seems to be correct.

mcuee, 2019-12-10:
Please create a pull request. Thanks.

commit 3f6e8baa (2020-04-02): Windows: Replace the sanitize_path() function with normalize_path()

Once upon a time the sanitize_path() function was needed to generate a
consistent path format in order to hash the resulting string for use as
session IDs. Since commit 71a779d078 ("Windows: Rework WinUSB
enumeration process to fix issues on Win8+"), this hashing method is no
longer used for session IDs, thus the sanitize_path() function was no
longer explicitly needed.

User lo1ol also reports in issue #633 that the function actually causes
issues with devices where there is a path component following the device
interface GU [... cut]
closed 2020-04-02 by dickens