libusb/libusb #641: I was able to edit the wikipage.

DavidBM, 2019-10-19:
I edited this one: https://github.com/libusb/libusb/wiki

Maybe there is permissions in order to avoid everyone to change things?

LudovicRousseau, 2019-10-19:
Why is it a problem to edit a wiki page?

DavidBM, 2019-10-19:
Because anyone can add malicious links, or delete everything, or add miss information.

pbatard, 2019-10-19:
That's hardly an issue. Malicious edits can be reverted (the history can __not__ be deleted, so even if someone deletes the whole thing, all it takes is about one click to restore it).

It's much better to let people edit a Wiki so that they can add relevant information, than close it off and slowly make it less and less relevant as it becomes obsolete.

If you see malicious links or misinformation, you can either add your edits to revert them or signal it to the project maintainers.

DavidBM, 2019-10-19:
Ok! 
The decision is yours. I will close this one. :)

closed 2019-10-19 by DavidBM