libusb/libusb #646: Clang LeakSanitizer report leak on macOs

vvhh2002, 2019-11-01:
I use a simple program to test libusb 1.0.23 with clang 8.0 on macOS 10.13.6.
But LeakSanitizer report error like this `
ASAN_OPTIONS=detect_leaks=1 ./clang_test
=================================================================
==76289==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 136 byte(s) in 1 object(s) allocated from:
    #0 0x1022ea612 in wrap_calloc (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x45612)
    #1 0x7fff5d4f6ce9 in class_createInstance (libobjc.A.dylib:x86_64h+0x6ce9)
    #2 0x7fff5e0e4c11 in _os_object_alloc_realized (libdispatch.dylib:x86_64+0x1c11)
    #3 0x7fff5e0e5f25 in dispatch_source_create (libdispatch.dylib:x86_64+0x2f25)
    #4 0x7fff5e102ade in _dispatch_kq_poll (libdispatch.dylib:x86_64+0x1fade)
    #5 0x7fff5e102741 in _dispatch_kq_drain (libdispatch.dylib:x86_64+0x1f741)
    #6 0x7fff5e1022dd in _dispatch_event_loop_poke (libdispatch.dylib:x86_64+0x1f2dd)
    #7 0x7fff5e0f5f65 in _dispatch_queue_resume_finalize_activation (libdispatch.dylib:x86_64+0x12f65)
    #8 0x7fff361ae26a in __CFRunLoopFindMode (CoreFoundation:x86_64h+0x5326a)
    #9 0x7fff361adec5 in __CFRunLoopCreate (CoreFoundation:x86_64h+0x52ec5)
    #10 0x7fff361adbc1 in _CFRunLoopGet0 (CoreFoundation:x86_64h+0x52bc1)
    #11 0x1021ebe00 in darwin_event_thread_main darwin_usb.c:486
    #12 0x7fff5e436660 in _pthread_body (libsystem_pthread.dylib:x86_64+0x3660)
    #13 0x7fff5e43650c in _pthread_start (libsystem_pthread.dylib:x86_64+0x350c)
    #14 0x7fff5e435bf8 in thread_start (libsystem_pthread.dylib:x86_64+0x2bf8)

Direct leak of 44 byte(s) in 1 object(s) allocated from:
    #0 0x1022ea23d in wrap_malloc (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x4523d)
    #1 0x7fff5e16c8b2 in _Block_copy (libsystem_blocks.dylib:x86_64+0x8b2)
    #2 0x7fff5e0e503b in _dispatch_Block_copy (libdispatch.dylib:x86_64+0x203b)
    #3 0x7fff5e0e60d8 in dispatch_source_set_event_handler (libdispatch.dylib:x86_64+0x30d8)
    #4 0x1022e9989 in wrap_dispatch_source_set_event_handler (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x44989)
    #5 0x7fff361ae23d in __CFRunLoopFindMode (CoreFoundation:x86_64h+0x5323d)
    #6 0x7fff361adec5 in __CFRunLoopCreate (CoreFoundation:x86_64h+0x52ec5)
    #7 0x7fff361adbc1 in _CFRunLoopGet0 (CoreFoundation:x86_64h+0x52bc1)
    #8 0x1021ebe00 in darwin_event_thread_main darwin_usb.c:486
    #9 0x7fff5e436660 in _pthread_body (libsystem_pthread.dylib:x86_64+0x3660)
    #10 0x7fff5e43650c in _pthread_start (libsystem_pthread.dylib:x86_64+0x350c)
    #11 0x7fff5e435bf8 in thread_start (libsystem_pthread.dylib:x86_64+0x2bf8)

Indirect leak of 64 byte(s) in 1 object(s) allocated from:
    #0 0x1022ea612 in wrap_calloc (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x45612)
    #1 0x7fff5e0e57ad in _dispatch_calloc (libdispatch.dylib:x86_64+0x27ad)
    #2 0x7fff5e10160c in _dispatch_unote_create (libdispatch.dylib:x86_64+0x1e60c)
    #3 0x7fff5e0e5f08 in dispatch_source_create (libdispatch.dylib:x86_64+0x2f08)
    #4 0x7fff5e102ade in _dispatch_kq_poll (libdispatch.dylib:x86_64+0x1fade)
    #5 0x7fff5e102741 in _dispatch_kq_drain (libdispatch.dylib:x86_64+0x1f741)
    #6 0x7fff5e1022dd in _dispatch_event_loop_poke (libdispatch.dylib:x86_64+0x1f2dd)
    #7 0x7fff5e0f5f65 in _dispatch_queue_resume_finalize_activation (libdispatch.dylib:x86_64+0x12f65)
    #8 0x7fff361ae26a in __CFRunLoopFindMode (CoreFoundation:x86_64h+0x5326a)
    #9 0x7fff361adec5 in __CFRunLoopCreate (CoreFoundation:x86_64h+0x52ec5)
    #10 0x7fff361adbc1 in _CFRunLoopGet0 (CoreFoundation:x86_64h+0x52bc1)
    #11 0x1021ebe00 in darwin_event_thread_main darwin_usb.c:486
    #12 0x7fff5e436660 in _pthread_body (libsystem_pthread.dylib:x86_64+0x3660)
    #13 0x7fff5e43650c in _pthread_start (libsystem_pthread.dylib:x86_64+0x350c)
    #14 0x7fff5e435bf8 in thread_start (libsystem_pthread.dylib:x86_64+0x2bf8)

Indirect leak of 40 byte(s) in 1 object(s) allocated from:
    #0 0x1022ea23d in wrap_malloc (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x4523d)
    #1 0x7fff5e16c8b2 in _Block_copy (libsystem_blocks.dylib:x86_64+0x8b2)
    #2 0x7fff5e16cb3d in _Block_object_assign (libsystem_blocks.dylib:x86_64+0xb3d)
    #3 0x7fff5e16c8ef in _Block_copy (libsystem_blocks.dylib:x86_64+0x8ef)
    #4 0x7fff5e0e503b in _dispatch_Block_copy (libdispatch.dylib:x86_64+0x203b)
    #5 0x7fff5e0e60d8 in dispatch_source_set_event_handler (libdispatch.dylib:x86_64+0x30d8)
    #6 0x1022e9989 in wrap_dispatch_source_set_event_handler (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x44989)
    #7 0x7fff361ae23d in __CFRunLoopFindMode (CoreFoundation:x86_64h+0x5323d)
    #8 0x7fff361adec5 in __CFRunLoopCreate (CoreFoundation:x86_64h+0x52ec5)
    #9 0x7fff361adbc1 in _CFRunLoopGet0 (CoreFoundation:x86_64h+0x52bc1)
    #10 0x1021ebe00 in darwin_event_thread_main darwin_usb.c:486
    #11 0x7fff5e436660 in _pthread_body (libsystem_pthread.dylib:x86_64+0x3660)
    #12 0x7fff5e43650c in _pthread_start (libsystem_pthread.dylib:x86_64+0x350c)
    #13 0x7fff5e435bf8 in thread_start (libsystem_pthread.dylib:x86_64+0x2bf8)

Indirect leak of 32 byte(s) in 1 object(s) allocated from:
    #0 0x1022ea23d in wrap_malloc (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x4523d)
    #1 0x7fff5e16cbbb in _Block_object_assign (libsystem_blocks.dylib:x86_64+0xbbb)
    #2 0x7fff5e16c8ef in _Block_copy (libsystem_blocks.dylib:x86_64+0x8ef)
    #3 0x7fff5e16cb3d in _Block_object_assign (libsystem_blocks.dylib:x86_64+0xb3d)
    #4 0x7fff5e16c8ef in _Block_copy (libsystem_blocks.dylib:x86_64+0x8ef)
    #5 0x7fff5e0e503b in _dispatch_Block_copy (libdispatch.dylib:x86_64+0x203b)
    #6 0x7fff5e0e60d8 in dispatch_source_set_event_handler (libdispatch.dylib:x86_64+0x30d8)
    #7 0x1022e9989 in wrap_dispatch_source_set_event_handler (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x44989)
    #8 0x7fff361ae23d in __CFRunLoopFindMode (CoreFoundation:x86_64h+0x5323d)
    #9 0x7fff361adec5 in __CFRunLoopCreate (CoreFoundation:x86_64h+0x52ec5)
    #10 0x7fff361adbc1 in _CFRunLoopGet0 (CoreFoundation:x86_64h+0x52bc1)
    #11 0x1021ebe00 in darwin_event_thread_main darwin_usb.c:486
    #12 0x7fff5e436660 in _pthread_body (libsystem_pthread.dylib:x86_64+0x3660)
    #13 0x7fff5e43650c in _pthread_start (libsystem_pthread.dylib:x86_64+0x350c)
    #14 0x7fff5e435bf8 in thread_start (libsystem_pthread.dylib:x86_64+0x2bf8)

SUMMARY: AddressSanitizer: 316 byte(s) leaked in 5 allocation(s).
`
my code like this:
`
extern "C"{
#include <libusb.h>
}
#include <cstdio>
#include <cstdlib>

int main(){
    printf("start!\n");
    libusb_device **devs;
    ssize_t cnt;
    libusb_init(nullptr);
    cnt = libusb_get_device_list(nullptr, &devs);
    if (cnt < 0)
        return (int)cnt;
    libusb_free_device_list(devs, 1);
    libusb_exit(nullptr);
    printf("end!\n");
    return 0;
}

`

LudovicRousseau, 2019-11-01:
Why can't I find the `start!` and `end!` in your output?

vvhh2002, 2019-11-02:
sorry output is incomplete.

> Why can't I find the `start!` and `end!` in your output?

LudovicRousseau, 2019-11-02:
The source of the problem looks to be this line https://github.com/libusb/libusb/blob/master/libusb/os/darwin_usb.c#L486
```
  runloop = CFRunLoopGetCurrent ();
  CFRetain (runloop);
```
The resource are correctly released at the end of the thread with line https://github.com/libusb/libusb/blob/master/libusb/os/darwin_usb.c#L556
```
  CFRelease (runloop);
```

So I don't know if/how we can fix the issue.

Do you have a patch to propose?

vvhh2002, 2019-11-04:
I think this may be a bug in Apple's runtime environment or a compiler configuration problem, I am trying to solve it.

seanm, 2020-03-20:
I took a look at that code, and don't see a leak, but of course sometimes they are hard to spot by inspection.

How many leaks does LSan find building say TextEdit?    https://developer.apple.com/library/archive/samplecode/TextEdit/Introduction/Intro.html   Maybe macOS is just leaky. :)

dickens, 2020-03-28:
I believe this to be a bug in the analyzer.  I've scoured the code and cannot see how this is a libusb issue.  With that said, I'm going to close this issue for now.  If new evidence comes to light, feel free to reopen this issue and post additional information.

closed 2020-03-28 by dickens