libusb/libusb #650: Best way to include the library on Android OS

Chakib-Temal, 2019-11-13:
hi

i followed all steps to include libusb on Android 

now i create an Android.mk and add the necessarys lines 
`include /home/chakib/Documents/NDKlibUsb/libusb-master/android/jni/libusb.mk

LOCAL_C_INCLUDES += $(LIBUSB_ROOT_ABS)
LOCAL_SHARED_LIBRARIES += libusb1.0`

the problem : 
when i start a new project C++ in Android , i have already a configuration with CmakeList.txt
with gradle : 
`externalNativeBuild {

        cmake {
            path "CMakeLists.txt"
        }
    }`

so when i add my external library libusb 
`externalNativeBuild {
        ndkBuild {
            path 'src/main/jni/Android.mk'
        }
    }`
it's cause an error for gradle : `Error:More than one externalNativeBuild path specified`

can we add the library in this file instead of Android.mk

or include the android.mk in the cmakeList.txt ? 

if yes ? how ? thank you

mcuee, 2019-12-07:
And this is not related to libusb, but rather related to CMake and android build. You can want to ask in CMake related mailing list.

mcuee, 2019-12-07:
Close for now.

mcuee, 2020-07-29:
Reopen this since this topic is mentioned in another ticket.

mcuee, 2021-02-05:
Close this for now.

closed 2021-02-05 by mcuee