libusb/libusb #651: unable to open device multiple times in windows QT project

bilsonzhao, 2019-11-18:
I use libusb-1.0.22 binary version in my windos qt project.  l also use Zadig to install WinUSB driver.
Now the problem is the first libusb_open operation can success,but the second time ,the libusb_open operation return LIBUSB_ERROR_ACCESS. I need to claim two interface ,so i must open the device twice,how to solve this problem.  Below is my code：

int r = 0;
    int i = 0;
    ssize_t cnt = libusb_get_device_list(context, &devs); 

    if (cnt <= 0)
    {
        qDebug() << "failed to get device list";
        return;
    }

    while ((dev = devs[i++]) != nullptr)
    {
        struct libusb_device_descriptor desc; 

        r = libusb_get_device_descriptor(dev, &desc); 
        if (r < 0)
        {
            qDebug() << "failed to get device descriptor"; //fail to get device descriptor
            return;
        }

        struct libusb_config_descriptor *config;
        r = libusb_get_config_descriptor(dev,0,&config);
        qDebug() << "interface number:" << config->bNumInterfaces;

        qDebug() << "vid = " << desc.idVendor << ",pid = " << desc.idProduct;
        if (desc.idVendor == MY_VID && desc.idProduct == MY_PID) 
        {

            r = libusb_open(dev, &CDCDATADEV); //打开设备，获取设备操作符
            if (r < 0)
            {
                qDebug() << "failed to open CDCDATADEV"; //fail to open device
                qDebug() << libusb_error_name(r) << " " << r;
                return;
            }

            r = libusb_open(dev, &VIDEODEV); 
            if (r < 0)
            {
                qDebug() << "failed to open VIDEODEV";
                qDebug() << libusb_error_name(r) << " " << r;
                return;
            } else {
                qDebug() << "success open VIDEODEV";
            }
            break;
        }
    }

    if (COMMONDEV != NULL)
    {
        if ((r = libusb_claim_interface(COMMONDEV, CDC_COMMOM_INTF)) < 0)
        { 

            r = libusb_detach_kernel_driver(COMMONDEV,CDC_COMMOM_INTF);
            if(r < 0) {
                qDebug() << "*** libusb_detach_kernel_driver failed! r = " << r;
            }

            r = libusb_claim_interface(COMMONDEV,CDC_COMMOM_INTF);
            if(r < 0)
            {
                qDebug() << "*** libusb_claim_interface failed! r = " << r;
                libusb_close(COMMONDEV);
                COMMONDEV = NULL;
            }

        }
    }

    if (CDCDATADEV != NULL)
    {
        if ((r = libusb_claim_interface(CDCDATADEV, CDC_DATA_INTF)) < 0)
        { 

            r = libusb_detach_kernel_driver(CDCDATADEV,CDC_DATA_INTF);
            if(r < 0) {
                qDebug() << "*** libusb_detach_kernel_driver failed! r = " << r;
            }

            r = libusb_claim_interface(CDCDATADEV,CDC_DATA_INTF);
            if(r < 0)
            {
                qDebug() << "*** libusb_claim_interface failed! r = " << r;
                libusb_close(CDCDATADEV);
                CDCDATADEV = NULL;
            }

        }
    }
    if (VIDEODEV != NULL)
    {
        // if ((r = libusb_set_configuration(VIDEODEV, MY_CONFIG)) < 0)
        // { //设置usb设备配置模式为1
        //     qDebug("error setting config #1: %s", libusb_error_name(r));
        //     libusb_close(VIDEODEV);
        //     VIDEODEV = NULL;
        // }

        if ((r = libusb_claim_interface(VIDEODEV, CDC_VIDEO_INTF)) < 0)
        { 
            r = libusb_detach_kernel_driver(VIDEODEV,CDC_VIDEO_INTF);
            if(r < 0) {
                qDebug() << "*** libusb_detach_kernel_driver failed! r = " << r;
            }

            r = libusb_claim_interface(VIDEODEV,CDC_VIDEO_INTF);
            if(r < 0)
            {
                qDebug() << "*** libusb_claim_interface failed! r = " << r;
                libusb_close(VIDEODEV);
                VIDEODEV = NULL;
            }

        }
    }

bilsonzhao, 2019-11-18:
CDCDATADEV can open success and works well ,but the VIDEODEV can not open and return LIBUSB_ERROR_ACCESS.

bilsonzhao, 2019-11-18:
this code in linux environment can works well,both CDCDATADEV and VIDEODEV can open success

diabolo38, 2019-11-20:
i suspect you installed one driver for the full device when you shall install multiple driver  of that composite device  , One for cdc (usualy 2 interface) and one for video .

You can also install libusbk back end instead of  winusb it should let you open multiple time the same device but that's quiet note the right way to solve your problem.

cristi-iacob, 2019-11-25:
Hi!

I have the same problem as @bilsonzhao. I try to simulate the hotplug mechanism for my device, getting the device list every 2 seconds and checking for new/old devices, but I'm receiving a LIBUSB_ERROR_ACCESS at the second attempt to open an interface for the device. This happens only on Windows. On Ubuntu 18 it works properly. 

I'm looking forward to a solution for this issue. 
Thanks!

bilsonzhao, 2019-12-05:
@diabolo38   my device is a composite deivice, the device have 5 inerface (three hid interface and one cdc command interface and one cdc data interface),even i install winusb driver 3 times for three hid interface,i can't open device multi times too.    

But when i use libusb-win32 and install driver for libusb-win32, i can open device multi times

mcuee, 2019-12-07:
@cristi-iacob Your issue is different. Take note Windows hotplug is not implemented and your device may take time to be enumerated by Windows.

mcuee, 2019-12-07:
@bilsonzhao You can use libusbk driver as the driver for the whole device (to replace the USB composite parent driver) and then try again.

And please post the descriptor here if that still does not work.

cristi-iacob, 2019-12-09:
> @cristi-iacob Your issue is different. Take note Windows hotplug is not implemented and your device may take time to be enumerated by Windows.

Thanks for the response, but i do not use hotplug. I just simulate the hotplug mechanism. Also, I don't think that the problem is the time Windows needs to enumerate the device, because I have let it process up to a few minutes, but it still couldn't open the device. More than this, it looks like libusb is finding the device but it's not able to open its interface for the second time.

mcuee, 2019-12-09:
@cristi-iacob please post the debug log.

RyanHope, 2020-05-25:
I am also getting this error in my projects where I try and simulate hotplug by polling the device list. Again, works perfectly fine in linux.

mcuee, 2020-06-02:
Please post the debug log.

mcuee, 2020-07-29:
@RyanHope Please post the debug log.

diabolo38, 2020-07-30:
Winusb backend want ever let you open or claim device multiple time
but for new device discovery I guess  you can make it w/o opening/claiming
Vid pid bus address or ports numbers shall be enough to sort out if it is new or same device but I can't remenber if that require opening device

mcuee, 2020-07-30:
https://github.com/libusb/libusb/wiki/Windows#Known_Restrictions

- Known Restrictions

- WinUSB cannot be used to send an actual reset command to an USB device. This is a limitation of WinUSB.

- WinUSB and libusbK cannot be used to set a device configuration that is different from the first one. This is a limitation of KMDF USB I/O Target.

- WinUSB does not support multiple concurrent applications (as per the Microsoft Windows Hardware Drivers documentation).

- WinUSB does not support isochronous transfers under Windows XP/Vista/7/8. WinUSB under Windows 8.1 or later supports isochronous transfer. libusb Windows supports isochronous transfer using the usbdk backend from version 1.0.21. libusb-1.0.22 adds isochronous support using libsubK driver. libusb-1.0.23 adds isochronous transfer support for WinUSB (Windows 8.1/10 or later).

mcuee, 2020-09-15:
Close this for now. Please reopen the ticket if you provide more info and you are not hitting the WinUSB limitation.

closed 2020-09-15 by mcuee