libusb/libusb #652: Include libraries in project after HomeBrow installation for Mac Os Users

Chakib-Temal, 2019-11-20:
this is a sample solution for OS X users , for those who have installed libusb and do not find it included in the standard library

if you follow the logs of hombrew, you will find the path that homebrew has chosen for the installation of the library on your mac

for my part, he added the library in "/ usr / local / Cellar / libusb",

well there is surely a way to add the library on the standard, but here I will show you an example of a cmakeList.txt file that you can use for your project so that it is available after 

open a new project and this is what you need to put in your cmakelist.txt file

**
cmake_minimum_required(VERSION 3.15)
project(libusb)

set(CMAKE_CXX_STANDARD 14)
add_executable(libusb main.cpp)

INCLUDE_DIRECTORIES(/usr/local/Cellar/libusb/1.0.23/include)
LINK_DIRECTORIES(/usr/local/Cellar/libusb/1.0.23/lib)

file(GLOB LIBRARIES "/usr/local/Cellar/libusb/1.0.23/lib/*.dylib")
message("LIBRARIES = ${LIBRARIES}")

TARGET_LINK_LIBRARIES(libusb  ${LIBRARIES})
**

restart your compilation and it should work

if you have an idea of ​​how I could add it back to the library that my gcc uses, you can post the solution down, thank you

LudovicRousseau, 2019-11-20:
You should use `pkg-config` to find the header directory and link flags instead of hardcoding the path. This would also work on other Unix systems, not just macOS with homebrew.

On my macOS with homebrew I have:
```
$ pkg-config --cflags libusb-1.0
-I/usr/local/Cellar/libusb/1.0.23/include/libusb-1.0
```
```
$ pkg-config --libs libusb-1.0
-L/usr/local/Cellar/libusb/1.0.23/lib -lusb-1.0
```

But I don't know how to use `pkg-config` with cmake. I let this as an exercice to the reader :-)

hjelmn, 2019-11-21:
Not a bug. There is a solution (pkg-config) so closing this.

closed 2019-11-21 by hjelmn