libusb/libusb #653: Simple example to start

Chakib-Temal, 2019-11-20:
Hi 

does you have a simple code to start ? i downloaded the library and i had compiled it 

how to start ? simple code with simple functions !  i have an HID device 

thank you

LudovicRousseau, 2019-11-20:
You should use the hidapi project for a HID device.
https://github.com/libusb/hidapi

mcuee, 2019-11-25:
Yes you should use HIDAPI instead.

However, if you really want to use libusb, here is an example (not for Mac OS X). Take note I am not a programmer myself.
https://github.com/mcuee/picusb/blob/master/lvr_genhid/libusb1_lvrhid4.c

closed 2019-11-25 by mcuee

Chakib-Temal, 2019-11-25:
thank you, i already found an sample example 

it's very simple to use 
https://www.dreamincode.net/forums/topic/148707-introduction-to-using-libusb-10/