libusb/libusb #654: libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access

duanbaihong, 2019-11-21:
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found
libusb: error [darwin_claim_interface] interface not found

hjelmn, 2019-11-21:
Error is self-explanatory. The device is in use by another program or a system driver.

Is this an HID device? If so please use hidapi instead.

mcuee, 2019-12-07:
Close for now. The OP can reopen if needed.

closed 2019-12-07 by mcuee