libusb/libusb #658: Unable to claim interface

Roman-Koshelev, 2019-12-03:
I use an example - xusb: Generic USB test program (xusb.c).  I'm trying to work with the keyboard.  It has 2 interfaces, one of which is not available (the one to which clicks come).  How can this be fixed?

clicks come to the address - 0x81.

program output:
Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 045E:07F8
         bcdDevice: 0300
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading first configuration descriptor:
             nb interfaces: 2
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.01
       endpoint[0].address: 81
           max packet size: 0008
          polling interval: 0A
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 82
           max packet size: 0003
          polling interval: 0A

Claiming interface 0...
   Failed. -12

Claiming interface 1...

Reading string descriptors:
   String (0x01): "Microsoft"
   String (0x02): "Wired Keyboard 600"

Roman-Koshelev, 2019-12-03:
(in Linux everything works fine out of the box)

pbatard, 2019-12-03:
https://github.com/libusb/libusb/wiki/Windows#Known_Restrictions

> HID keyboards and mice cannot be accessed using the native HID driver as Windows reserves exclusive access to them.

closed 2019-12-03 by pbatard