libusb/libusb #660: hotplug example not working on macOS Mojave

kakyoism, 2019-12-06:
I tried to run the libusb sample code on macOS Majave

https://github.com/libusb/libusb/blob/master/examples/hotplugtest.c

only to find that the callbacks are not fired up when hotplugging in my iPhone to my Mac.

Is this a known issue?

mcuee, 2019-12-10:
Please post the debug log.

jhihn, 2019-12-23:
I am on Mojave and hotplug works fine for me.

hisschemoller, 2019-12-30:
I have the same as @kakyoism ,  hotplugtest runs without error but no callbacks fired when I plug or unplug several USB devices. Two other examples, listdevs and testlibusb run fine.

Tried on two Macbooks with the same result. Macbook pro 2018, MacOS Mojave and Macbook pro 2010, OSX El Capitan. Libusb installed with Homebrew in both cases.

I'm just starting with c programming so I have not much knowledge. I don't know how to generate a debug log yet.

dickens, 2020-01-13:
How are you invoking the hotplug example?  You must provide the VID and PID of the device you are interested in as command line arguments, otherwise it defaults to searching for VID 0x045a and PID 0x5005, which I believe is a Rio S10 MP3 player.

kakyoism, 2020-01-14:
@dickens That's a surprise. Then how could I detect hogplugs from ANY USB devices?

dickens, 2020-01-14:
> @dickens That's a surprise. Then how could I detect hogplugs from ANY USB devices?

The hotplug example would have to be enhanced to do that.  The example program is really meant to verify that hotplug works on your system given a specific device that you will be plugging in and out.

seanm, 2020-06-25:
Seems like this ticket could be closed...

closed 2020-07-29 by mcuee