libusb/libusb #661: libusb: error,another process has device opened for exclusive access

LAIhy, 2019-12-10:
I try build a remote desktop client on macOS, the system architecture is virt-viewer + spice-gtk + libusb. In order to achieve USB devices redirection, I try to access USB devices through libusb, but I met a problem,
the log is：
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access

I have tried the methods described in the FAQ，but it's still not working
1, sudo kextunload -b com.apple.driver.AppleUSBFTDI,
and I got the return: com.apple.driver.AppleUSBFTDI not found for unload request. Failed to unload com.apple.driver.AppleUSBFTDI 

2, sudo kextunload FTDIUSBSerialDriver.kext,
and I got: Unable to determine realpath for /FTDIUSBSerialDriver.kext - failing. Can't create FTDIUSBSerialDriver.kext

the libusb version I used is 1.0.23,
the macOS version is 10.14.6 Mojave

Is there any other ways to fix this problem?

mcuee, 2019-12-10:
Please use libusb mailing list for technical support. Github is more for anomaly reporting and enhancement request or pull requests.

mcuee, 2019-12-10:
Anyway, what is your device? Why do you think it is related to FTDI driver?

LAIhy, 2019-12-11:
A USB flash disk actually, I am not quite sure which driver is related, But I think it is more likely related to the IOUSBMassStorageDriver?, I have try to unload this driver too, and I got :

(kernel) Can't remove kext com.apple.iokit.IOUSBMassStorageDriver; services failed to terminate - 0xdc008018.
Failed to unload com.apple.iokit.IOUSBMassStorageDriver - (libkern/kext) kext is in use or retained (cannot unload).

I am not  familiar with macOS kernel, so I try to unload FTDI driver that described in the FAQ : https://github.com/libusb/libusb/wiki/FAQ

And I didn't know the mailing list before , I will post my question there , thank

mcuee, 2019-12-11:
Close this. Please do not use a flash disk with libusb under any OS, especially macOS.

closed 2019-12-11 by mcuee

huisetouxiang, 2020-11-03:
> Close this. Please do not use a flash disk with libusb under any OS, especially macOS.

Hello, have you solved this problem？

huisetouxiang, 2020-11-03:
> I try build a remote desktop client on macOS, the system architecture is virt-viewer + spice-gtk + libusb. In order to achieve USB devices redirection, I try to access USB devices through libusb, but I met a problem, I plug in a USB mass storage device, and I got the error log like this：
> libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
> 
> I have tried the methods described in the FAQ，but it's still not working
> 1, sudo kextunload -b com.apple.driver.AppleUSBFTDI,
> and I got the return: com.apple.driver.AppleUSBFTDI not found for unload request. Failed to unload com.apple.driver.AppleUSBFTDI
> 
> 2, sudo kextunload FTDIUSBSerialDriver.kext,
> and I got: Unable to determine realpath for /FTDIUSBSerialDriver.kext - failing. Can't create FTDIUSBSerialDriver.kext
> 
> the libusb version I used is 1.0.23,
> the macOS version is 10.14.6 Mojave
> 
> Is there any other ways to fix this problem?

Hello, have you solved this problem？