libusb/libusb #663: Python libusb gets access violation listing devices

enjoysmath, 2019-12-11:
[image]

Minimal code:
```
import libusb as usb
from ctypes import POINTER, c_void_p

class USBDriver:
   def __init__(self, ctx:usb.context=None):
      self._context = None
      self.context = ctx      

   @property
   def context(self):
      return self._context
   
   @context.setter
   def context(self, ctx=None):
      if ctx != self._context:
         self.exit_usb_library()
         self._context = ctx
         res = usb.init(ctx)
         if res < 0:
            raise Exception("Error initializing the libusb Python binding. ctx=", ctx)         
         if __debug__:
            usb.set_debug(ctx, 3)   # Set verbosity level                 
         
   def get_devices(self):
      devices = POINTER(POINTER(POINTER(usb.device)))()
      usb.device_descriptor
      count = usb.get_device_list(self.context, devices)
      print(devices)
         
   def exit_usb_library(self):
      usb.exit(self.context)
      
      
      
if __name__ == '__main__':
   
   driver = USBDriver()
   driver.get_devices()
```
As you can see to get LP_LP_LP_device I used ctypes.  The next line throws the violation.

LudovicRousseau, 2019-12-11:
Why do you think it is a problem with libusb and not a problem the Python wrapper?

mcuee, 2019-12-11:
@enjoysmath why reinvent the wheel? Why not use pyusb?

mcuee, 2019-12-11:
Close for now. You can ask in pyusb mailing list if you want.

closed 2019-12-11 by mcuee