libusb/libusb #667: Control Transfer function not returning the actual length returned by the DATA packet

GarageGadget, 2019-12-23:
Hi,

The Control Transfer function (sync) [_libusb_control_transfer_] not returning the actual length returned by the DATA packet but instead returning the size of the assigned buffer.

The issue appears to be the result of the _ControlTransfer_ returning True at which point the overlapped structure contains _overlapped->InternalHigh_ with the correct number of data bytes. 
The _windows_force_sync_completion_ function is called with the buffer size which is then used to set the _overlapped->InternalHigh_ to the buffer size overriding the actual received bytes.

I used _Wireshark_ to verify the actual data  size returned by the Overlapped structure by the ControlTransfer is correct and not the buffer size.

**Example:**
My application is expecting 2 bytes but the _libusb_control_transfer_ returns 8 bytes which is the size of the buffer passed into the libusb_control_transfer.

mcuee, 2020-01-03:
Sorry but just wondering if you see the same thing under other OS like macOS or Linux if you have access to those OS.

GarageGadget, 2020-01-03:
Hi,

 

The issue appears to be isolated to the Windows OS and is working on Raspbian without issue.

 

BR

dickens, 2020-01-17:
Thanks for the report.  Fix coming soon.

dickens, 2020-02-05:
Reopening as this isn’t quite fixed yet.  I’ll close it again with the proper fix.

dickens, 2020-02-07:
This should now _really_ be fixed in 9c28ad219b654011783a42ec888ca87dbda704a6

commit ce95f656 (2020-01-28): Windows: Fix reported length of synchronous control transfers

WinUSB control transfers that complete synchronously are incorrectly
having the actual transfer length set to the size of the transfer
buffer. If the control transfer is a read, the device may return less
data than the transfer buffer size. Fix this by reporting the actual
bytes transferred.

Closes #667
closed 2020-02-07 by dickens