libusb/libusb #672: Maximum Transfer Size for bulk In transfer

winkingwolf, 2019-12-30:
hi,
I use bulk transfer on winusb/win7, I use fill-submit-LIBUSB_CALL mode for bulk transfer. The biggest size of transfer seems to be 2MB, I want to transfer 5MB in one transfer,ONLY 2MB is in good transfer, the other part（5MB-2MB）is not in good transfer.

Is it because the limitation of WinUSB driver or something else?
What is the Maximum Transfer Size for bulk  transfer for Libusb?
OS:win7/winusb

mcuee, 2020-01-03:
I believe that is a WinUSB driver limitation. It is said the limit is 2.1MB.

Ref: https://community.osr.com/discussion/189326/changing-maximum-transfer-size-pipe-policy-in-winusb

mcuee, 2020-01-07:
Close this for now.

closed 2020-01-07 by mcuee