libusb/libusb #675: [Question] Can I send only setup packet for control transfer?

MrKris99, 2020-01-12:
Hi! I just got a situation when I need to violate usb protocol by sending only setup packet to device (so I need to skip data stage).  I can achive it by creating asynchronius transfer and immediatly cancel it by it's really unstable. Can I have such low level conrol with libusb?

mcuee, 2020-01-13:
Not possible.

MrKris99, 2020-01-13:
@mcuee Maybe I can achieve it by patching libusb?

mcuee, 2020-01-13:
Not possible as this is by definition of control transfer and in the HW level and low level OS driver level.

mcuee, 2020-01-13:
What is your end goal?

MrKris99, 2020-01-13:
I'm trying to research some vulnerabilities and need to have full control of Usb stack. I've got success with the use Arduino controller and it will by nice to port it for Windows or Linux.

mcuee, 2020-01-13:
That is below libusb. I believe you are using the non-USB-Spec-Conforming low speed only FW USB implementation of AVR. In that case you have control of the behavior. 

What you did is on the device side, not possible on the Host side -- I think the host will reject your device.

libusb is on the host side.

mcuee, 2020-01-13:
Please continue further discussion on libusb mailing list. Thanks. I think this is not related to libusb but relate to general understanding of USB. It is okay to discuss such questions in the mailing list.

closed 2020-01-13 by mcuee