libusb/libusb #677: How can I run libusb applications under Mac OS X if there is already a kernel extension installed for the device? On Catalina

aloserini, 2020-01-17:
How can I run libusb applications under Mac OS X if there is already a kernel extension installed for the device? On Catalina

Because under Catalina AppleUSBFTDI.kext doesn't exist anymore.

I have this Problem:
Platform does not support kernel detachment
Unable to claim USB device interface

LudovicRousseau, 2020-01-17:
If there is a kernel extension for your device why don't you use the kernel driver?
What is your device?

aloserini, 2020-01-17:
I have a Hackintosh with a Corsair Fan Controller, plugged in de Motherboards USB.
My Mac OS Catalina detect this Controller as a UPS (Uninterruptible Power Supply)
Now I need to detach this Kernel Driver to use the USB with another Script.
Can you Help me?
The Script: https://github.com/audiohacked/OpenCorsairLink

LudovicRousseau, 2020-01-17:
No sorry. I never detached kernel drivers.

mcuee, 2020-01-20:
This is basically not possible under new version of macOS with inbox Apple kernel driver. Unless you are a big customer of Apple and you want to use your own driver. 

Last time it was possible with codeless kext but it is no longer possible.

mcuee, 2020-01-20:
Ref: you need to get a kext signing key from Apple
https://forums.developer.apple.com/thread/95184
Ref: kext is deprecated
https://forums.developer.apple.com/thread/120246

mcuee, 2020-01-20:
Your device is not using FTDI vendor driver as it is not a FTDI device. It seems to me some of the device supported by the OpenCorsairLink is HID device, in that case you should use HIDAPI instead.

Please tell us the kernel driver used.

closed 2020-01-20 by mcuee