libusb/libusb #679: libusb_bulk_transfer err ( return -5) on win7_64

lfjking, 2020-01-20:
On my progress , using libusb_get_device_list  ,libusb_open_device_with_vid_pid,  libusb_claim_interface,  libusb_set_interface_alt_setting is ok.  when I libusb_bulk_transfer write or read, return -5 ("/** Entity not found */ LIBUSB_ERROR_NOT_FOUND = -5,"). I don't know how to do. I want to use my progress win / linux . so I need libusb. 
help, please.

lfjking, 2020-01-20:
my libusb ver is 1.0.22

dickens, 2020-01-21:
I think the only reason you would get this is if the endpoint specified for the transfer doesn't belong to a claimed interface.  Can you double-check this?  You can also set LIBUSB_DEBUG=4 in the environment to enable debug messages.

lfjking, 2020-01-22:
Thank you very much, I will try to test this problem.

lfjking, 2020-01-23:
> 
> 
> I think the only reason you would get this is if the endpoint specified for the transfer doesn't belong to a claimed interface. Can you double-check this? You can also set LIBUSB_DEBUG=4 in the environment to enable debug messages.

I created a new engineering test libusb.
I found that it was my mistake.
But there is a new problem.
my device is android, one interface had two endpoints. now the endpoint_out is ok, but the endpoint_in return -1(the error is Input/output error). 
I tested it for a long time, but I couldn't find the reason.

lfjking, 2020-01-25:
`									int relen = 0;
									libusb_device_handle_ptr mLibHandle = NULL;
									libusb_open(dev, &mLibHandle);
									if (!mLibHandle){
										return ret;
									}

									//setting
									if (libusb_set_configuration(mLibHandle, _config->bConfigurationValue) != LIBUSB_SUCCESS){ return ret; }
									if (libusb_claim_interface(mLibHandle, _config->bInterfaceNumberValue) != LIBUSB_SUCCESS){ return ret; }
									if (libusb_set_interface_alt_setting(mLibHandle, _config->bInterfaceNumberValue, _config->bAlternateSettingValue) != LIBUSB_SUCCESS){ return ret; }
									unsigned char buff[4] = { 0 };
									libusb_bulk_transfer(mLibHandle, _config->bEndpoint_Out, (unsigned char*)"adb", 3, &relen, NULL);
									libusb_bulk_transfer(mLibHandle, _config->bEndpoint_In, buff, 1, &relen, 2000);**### //here is return -1**
									libusb_release_interface(mLibHandle, _config->bInterfaceNumberValue);
									libusb_close(mLibHandle);
									//-------------------test-------------------------------`

closed 2020-01-26 by lfjking